Rodemfr/MicronetToNMEA #2: No Networks Found

mattnj, 2021-09-13:
Hello!
I have build this circuit with a Teensy 3.5 and CC1101 (i have tried 2 different CC1101 boards) but it does not detect any Micronet Networks, I have tried onboard and nothing was detected (T121, T120 with T909 compass a wind, MN100 and MN100-2)

I now have at home on bench with just a MN100-2 unit, should that be detected?

What are the steps to debug/check this please? Any pointers? Excited to get it working!

Many Thanks

*** MicronetToNMEA ***

0 - Print this menu
1 - General info on MicronetToNMEA
2 - Scan Micronet networks
3 - Attach converter to a network
4 - Start NMEA conversion
5 - Scan all surrounding Micronet traffic
6 - Calibrate magnetometer
7 - Test heading
Choice : 1

MicronetToNMEA, Version 0.4
Device ID : 3123456
No Micronet Network attached
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
No navigation compass detected, disabling magnetic heading.
Provides the following NMEA sentences :
 - INDPT (Depth below transducer. T121 with depth sounder required)
 - INMWV (Apparent wind. T120 required)
 - INMWV (True wind. T120 and T121 with Speedo/Temp sensor required)
 - INMTW (Water temperature. T121 with Speedo/Temp sensor required)
 - INVHW (Speed on water. T121 with Speedo/Temp sensor required)
 - INVLW (Distance log T121 with Speedo/Temp sensor required)

Choice : 2

Scanning Micronet networks for 5 seconds ... done

/!\ No Micronet network found /!\
Check that your Micronet network is powered on.

Rodemfr, 2021-09-14:
Hi mattjn,

I have no simple direct explanation of your problem. We have to investigate/debug. I read on YBW forum that you have compiled with Arduino IDE instead of Sloeber. In theory, this is not an issue, but who knows ? So, if you give me your pin configuration (if it is different from the one suggested in MicronetToNMEA front page), I could provide you an executable compiled by myself so that we could check that you compilation environment is OK. Send me a private essage with you email so that we can exchange binaries.Another track to follow is the interrupt line (GD0) from CC1101 to Teensy. If you can, check is this line is toggling when using menu 5.
In general, I use menu 5 to see if there is a surrounding network.

Best regards,

Ronan.

mattnj, 2021-09-15:
I have just checked and GD0 > Pin24 is not doing anything when menu5 is selected.

dwarning, 2021-09-15:
How do you checked GDO0? With an oscilloscope? If yes you could build simpler test configurations and check with one of the SmartRC lib examples if your teensy controls the SPI bus.
Can you post a picture of your hardware setup. Do you have same CC1101 board like me?

mattnj, 2021-09-15:
yes with oscilloscope, i will try an example, I have tried 2 different CC1101 as pictured.

[image]
[image]

dwarning, 2021-09-15:
I made checks with the blue one, but not successful. It seems a 433MHz board and show very low transmit power with 869MHz. I use the green board,
The LED is on - something is wrong. For me it was the wrong I2C connection as this LED lightning.
I would not start with the Bluetooth Serial. Remove it so far not anything is OK. 
Your wiring looks dangerous. Take care on the lower left side - your Vin is very tight to the USB ground.
What is this right from teensy board? Magnetometer?
What is this grey wire in the middle near the SD slot?
I started with these universal breadboards.
[image]

You can not solder so often on the teensy board.
Please simplify your setup.

mattnj, 2021-09-15:
Thanks, yes, I soldered it perfectly 1st time with the green board, then have pulled apart and now its a bit of a mess! :-( 
I have ordered another teensy and will start again.

dwarning, 2021-09-15:
OK, but use teensy with pins. And use adapter or breadboard.
You not answered my other questions.

mattnj, 2021-09-15:
What is this right from teensy board? that's an ESP8266 to send data over Wifi (that bit works!)
What is this grey wire in the middle near the SD slot? - Just a GND wire

Rodemfr, 2021-09-15:
Hi mattnj,

I sent a working HEX file on your email.
If this does not solve your issue, I have tracks to investigate :

You reached menu on the console. It means that your cc1101 was successfully detected on I2C by the Teensy. However, you don't receive interrupts from GD0 pin of CC1101. It means that CC1101 doesn't catch packets from you Micronet devices. I see two possible explanations : 
- Your oscillator is not at the same frequency than ours. Each manufacturer can choose different oscillator on his board as long as it is in CC1101's spec. We all buy cheap breakout board on internet which are often low quality products. It might be that MicronetToNMEA is not configured properly for your board by default. In this case, it means that we have to find the right settings for your frequency
- Second possible option : are you American ? If yes, I read somewhere that the frequency of Micronet devices is shifted toward 900MHz instead of 868MHz in the US. It might explain the issue.

In all cases, we will probably enter in a more complex debug phase to findout what is the problem. Have you something like a RTL-SDR dongle to scan Micronet wireless communications ? It would help understanding what happens.

mattnj, 2021-09-15:
Hello. I'm in UK...and devices are 868mhz....I do have a rtl-sdr I just bought and it does appear to be receiving on that frequency although I'm.not sure exactly what I'm looking at....I will post images tomorrow of scan.

dwarning, 2021-09-15:
OK, that's sound fine.
First switch only the display on. Then look with SDR# or adequate software to get a picture similar like that.
[image]
Then take the attached hello world file and load it to the teensy. 
[cc1101_Transmitt_Hello_World_advanced.zip](https://github.com/Rodemfr/MicronetToNMEA/files/7172984/cc1101_Transmitt_Hello_World_advanced.zip)
Then switch only the teensy + CC1101 on and look if can see a similar spectrum.

mattnj, 2021-09-17:
[image]

This is what I see on SDR, there appears to be a lot of activity when turned on for about 5 secs, then it settles into a blip once a second.

dwarning, 2021-09-17:
OK,
if the display switched on, it is in kind boot phase and established the network. It is not of interest.
What you need is the continuous 1s bursts. If you see them in URH spectrum analyzer you can try to catch them with the recorder. How it should look you can see in the main YBW thread. You need the 555555...5599XXXXXXXX
[image]
The four bytes after 99 are your networkID. It is bit fiddling around with all the URH features. Sometime smoothing helps. Or shifting the center. Or cutting the settling phase as long you have 5599XXXXXXXXXXXXXX.
Please comment out the debug include in the hello world .ino file. Then start it again (Raymarine Display off!) and look for the spectrum. It should same frequency like you posted for the display. If not you must bring the frequency setting in line. The script is only to check the transmit capability of your teensy-cc1101 combination.

m-karee, 2021-09-18:
> Hello!
> 
> I have build this circuit with a Teensy 3.5 and CC1101 (i have tried 2 different CC1101 boards) but it does not detect any Micronet Networks, I have tried onboard and nothing was detected (T121, T120 with T909 compass a wind, MN100 and MN100-2)
> 
> 
> 
> I now have at home on bench with just a MN100-2 unit, should that be detected?
> 
> 
> 
> What are the steps to debug/check this please? Any pointers? Excited to get it working!
> 
> 
> 
> Edit* I bought a SDR radio and with one tacktick unit powered on, it does appear to be doing something on the frequency on the SDR.... Nothing received on Cc1101/teensy console...
> 
> 
> 
> Is there a debug mode?!
> 
> 
> 
> Many Thanks
> 
> 
> 
> 
> 
> 
> 
> *** MicronetToNMEA ***
> 
> 
> 
> 0 - Print this menu
> 
> 1 - General info on MicronetToNMEA
> 
> 2 - Scan Micronet networks
> 
> 3 - Attach converter to a network
> 
> 4 - Start NMEA conversion
> 
> 5 - Scan all surrounding Micronet traffic
> 
> 6 - Calibrate magnetometer
> 
> 7 - Test heading
> 
> Choice : 1
> 
> 
> 
> MicronetToNMEA, Version 0.4
> 
> Device ID : 3123456
> 
> No Micronet Network attached
> 
> Wind speed factor = 1.00
> 
> Wind direction offset = 0
> 
> Water speed factor = 1.00
> 
> Water temperature offset = 0
> 
> No navigation compass detected, disabling magnetic heading.
> 
> Provides the following NMEA sentences :
> 
>  - INDPT (Depth below transducer. T121 with depth sounder required)
> 
>  - INMWV (Apparent wind. T120 required)
> 
>  - INMWV (True wind. T120 and T121 with Speedo/Temp sensor required)
> 
>  - INMTW (Water temperature. T121 with Speedo/Temp sensor required)
> 
>  - INVHW (Speed on water. T121 with Speedo/Temp sensor required)
> 
>  - INVLW (Distance log T121 with Speedo/Temp sensor required)
> 
> 
> 
> Choice : 2
> 
> 
> 
> Scanning Micronet networks for 5 seconds ... done
> 
> 
> 
> /!\ No Micronet network found /!\
> 
> Check that your Micronet network is powered on.
> 
> 
> 
>

dwarning, 2021-09-18:
@mattnj :
The attached receive script shows the address of your network in first 4 bytes:
[cc1101_Receive_advanced.zip](https://github.com/Rodemfr/MicronetToNMEA/files/7190178/cc1101_Receive_advanced.zip)
You must switch on only the display.
Hope it helps.

Rodemfr, 2021-09-20:
Hi mattnj,
dwarning just found an issue with CC1101 which could explain your troubles. I just updated master branch with the fix. You should probably try this new version.

mattnj, 2021-09-21:
> Hi mattnj,
> dwarning just found an issue with CC1101 which could explain your troubles. I just updated master branch with the fix. You should probably try this new version.

YES! i dint need to do anything other than run the new code! Thank you!

Just a small point with USB console you cant see any output when connected and booted up....you need to add a delay(250) after initialising serial ports I added this:-

	// Init USB serial link
	USB_CONSOLE.begin(USB_BAUDRATE);

	// Init NMEA GNSS serial link
	GNSS_SERIAL.setRX(GNSS_RX_PIN);
	GNSS_SERIAL.setTX(GNSS_TX_PIN);
	GNSS_SERIAL.begin(GNSS_BAUDRATE);

	// Init serial link with HC-06
	BLU_CONSOLE.setRX(BLU_RX_PIN);
	BLU_CONSOLE.setTX(BLU_TX_PIN);
	BLU_CONSOLE.begin(BLU_BAUDRATE);

//wait for Serials to start up then show menu
  delay (250);

	// Setup main menu
	gMenuManager.SetMenu(mainMenu);

dwarning, 2021-09-21:
Fine that it runs.
To your request:
`  Serial.begin(115200);
  while (!Serial); `
should also do the job.

closed 2021-09-21 by dwarning