Rodemfr/MicronetToNMEA #3: esp32-trial - does this branch work?

j-lang, 2022-03-28:
Thanks for this great development. 
Because a ESP32 board ist available and also a CS1101 868MHz board, I tried to compile the esp32-trial branch. 
Due to the notice in the documentation, that there are some issues with the esp32 implementation, I want to know, is the ESP32- branch will work? 
After building the code there are most of the time CS1101 not available errors and the board has to be rebooted. 
Therefore it would be great to know, if the code should work in general, or it absolutely necessary to buy a Teensy board?

dwarning, 2022-04-11:
No, the branch did not work. It was an experiment.

Teensy board is preferable.

j-lang, 2022-04-15:
Thanks, I got in the meantime a Teensy board and it works.

closed 2022-04-15 by j-lang