Rodemfr/MicronetToNMEA #4: Info: protocol issue solved by 33d0859

j-lang, 2022-04-10:
**Hardware**
Teensy 4.1
CC1101
uBlox M8N GPS without configuration flash memory

**Problem**
Up to commit [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5)
my T110 Display was not able to show GPS coordinates when RMC messages was activated to update time and date. When only the GGA message was activated, the position data was shown. (I have a compass that I does not need the VTG message)

With commit [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5)
GPS position and time and date is shown with activated GGA and RMC message. 

I am not familiar with GIT. I made a simple extension for setting up uBlox GPS modules without configuration flash memory. 
How can I send you these code parts?

dwarning, 2022-04-11:
To be honest, I couldn't understand the problem. Commit [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5) is the last commit on master from @Rodemfr. You say with this commit position and time is shown. Then would be anything OK. Or?
You are sure that your module has not flash? I thought it too as I got my module first. Then by playing around with U-Center to change baudrate I found out it has. Which module you have? Photo or link.
In my opinion it makes no sense to disable certain NMEA messages. Fix time is not impacted. What is your reason for deactivating messages?
Without git you can make a patch by diff the code. And we can see if and how we can include this. Your extension is related to baudrate or message type disabling?

j-lang, 2022-04-15:
I rechecked the problems: there are no differences between [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5) and the version before.

The following combination of active messages where shown on the T110 display - after changing the message settings, the display was turned off and on:
GGA + RMC, GGA + VTG, RMC + VTG.
If GGA + RMC + VTG is activated, the display does not show any info (time & date, COG, lat/lon) 

I rechecked the problems: there are no differences between [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5) and the version before.

The following combination of active messages where shown on the T110 display - after changing the message settings, the display was turned off and on:
GGA + RMC, GGA + VTG, RMC + VTG.
If GGA + RMC + VTG is activated, the display does not show any info (time & date, COG, lat/lon) 

I rechecked the problems: there are no differences between [33d0859](https://github.com/Rodemfr/MicronetToNMEA/commit/33d0859cc0aae0bb0a7309ee4549b19f30b5aba5) and the version before. 

The following combination of active NMEA messages where shown on the T110 display - after changing the message settings, the display was turned off and on:
GGA + RMC, GGA + VTG, RMC + VTG.
If GGA + RMC + VTG is activated, the display does not show any info (time & date, COG, lat/lon), the same after a gps module reset, when all message are active. -> that is the reason for deactivating messages. 
The ublox module (photo attached) seems to be a cheap module that does not allow to store settings. 

The code part for GPS are changing the baud rate and deactivate unnecessary messages. 

[image]
``

j-lang, 2022-04-17:
I received the compass module and attached it. The HDG display shows immediately a value but the NMEA messages from GPS are not shown since active compass. If the compass is disconnected and the display ist restarted, COG, lon/lat or time is shown.

j-lang, 2022-04-29:
Comited GPS initalization code that allows to configure this cheap ublox that starts with 9600bps after power up. 
some NMEA messages can be enabled and disabled.

j-lang, 2022-04-30:
May there be an issue with the Teensy 4.1 (4.0) board, when the software runs perfect with the Teensy 3.5?

dwarning, 2022-05-01:
It's not easy to come in conversation with you, but anyway:
Do you miss the messages in the serial console (F4) and on display or only on display?
Can you grep your serial output and show us.
Did you follow my proposal and try u-center configuration and spotting the serial interface for the missed messages?

j-lang, 2022-05-01:
Hi dwarning,
_**Do you miss the messages in the serial console (key 4) and on display or only on display?**_
I miss info in display, but messages are shown in the console.
_**Can you grep your serial output and show us.**_
Yes! Sample output with GGA and VTG messages activated on the GPS*), the display shows HDG, COG, SOG, lat, lon: 
```
$GNVTG,,T,,M,0.183,N,0.340,K,A*30
$GNGGA,092755.00,46xx.x3209,N,01xxx.x3677,E,1,06,1.40,545.4,M,43.4,M,,*48
$GNVTG,,T,,M,0.145,N,0.269,K,A*30
$GNGGA,092756.00,46xx.x3198,N,01xxx.x3666,E,1,06,1.40,545.3,M,43.4,M,,*47
$INMWV,18.0,R,0.0,N,A*03
$GNVTG,,T,,M,0.310,N,0.574,K,A*39
$GNGGA,092757.00,46xx.x3199,N,01xxx.x3661,E,1,06,1.40,544.8,M,43.4,M,,*4A
$GNVTG,,T,,M,0.429,N,0.795,K,A*39
$GNGGA,092758.00,46xx.x3197,N,01xxx.x3649,E,1,06,1.40,544.8,M,43.4,M,,*41
```
Sample output with GGA, VTG and RMC messages activates on the GPS, the display shows HDG, but COG, SOG, lat, lon showing '---'
```
$INMWV,18.0,R,0.0,N,A*03
$GNRMC,093154.00,A,46xx.x3618,N,01xxx.x3814,E,0.329,,010522,,,A*61
$GNVTG,,T,,M,0.329,N,0.610,K,A*32
$GNGGA,093154.00,46xx.x3618,N,01xxx.x3814,E,1,07,1.58,556.5,M,43.4,M,,*4A
$GNRMC,093155.00,A,46xx.x3632,N,01xxx.x3779,E,0.086,,010522,,,A*6A
$GNVTG,,T,,M,0.086,N,0.160,K,A*34
$GNGGA,093155.00,46xx.x3632,N,01xxx.x3779,E,1,07,1.58,556.8,M,43.4,M,,*4A
$INMWV,18.0,R,0.0,N,A*03
$GNRMC,093156.00,A,46xx.x3659,N,01xxx.x3762,E,0.398,,010522,,,A*62
$GNVTG,,T,,M,0.398,N,0.738,K,A*33
$GNGGA,093156.00,46xx.x3659,N,01xxx.x3762,E,1,07,1.58,555.8,M,43.4,M,,*4D
$GNRMC,093157.00,A,46xx.x3693,N,01xxx.x3740,E,0.207,,010522,,,A*62
$GNVTG,,T,,M,0.207,N,0.384,K,A*37
$GNGGA,093157.00,46xx.x3693,N,01xxx.x3740,E,1,08,1.20,556.2,M,43.4,M,,*43
$INMWV,18.0,R,0.0,N,A*03
```
*) the GPS messages can be easily be activated with my initialization code of my pull request. 
Deactivation by simply commenting unwanted message types in constants part. 

_**Did you follow my proposal and try u-center configuration and spotting the serial interface for the missed messages?**_
I can configure and save configuration, but after disconnecting the power supply the configuration is lost. 
After power loss the module sends always all NMEA message types.  
Maybe this post is also appropriate for my module: [How to save the configuration for M8N GPS?](https://portal.u-blox.com/s/question/0D52p0000AEWU5ICQX/how-to-save-the-configuration-for-m8n-gps)
Therefore I added the configuration of GPS to the code, where the GPS module gets after power up the decided config (pull request) 

**My observations and misinterpretations**
- the T110 and T112 are showing received infos for a while and if there is no update they are showin '---' -> therefore now I am switch them off an on after change on the code and disconnect the also the teensy from power supply to get a stable test condition.
- GGA is necessary for lat, lon
- VTG is necessary for COG, SOG
- RTM is necessary for time and date on T110
- If I want time and date only RTM + GGA without HDG (compass disconnected) or RTM + VTG without HDG (compass disconnected) or RTM with HDG (compass connected) possible. 
In my first posts, there are some observations wrong because the the T110 display is showing received messages and it takes a while till they change to '---' when they are not received anymore.. 

This GPS module is only for laboratory and testing, before I build the final part for the boat. 
Therefore I started with ESP32, CC1101 and a GPS module available in my workshop. The teensy 3.5 cannot be ordered anywhere and it took a while to check if the teensy 4.0/4.1 can be used for this project.

davidwkerr, 2022-05-03:
I thought I read somewhere that Rodemfr went sailing with the code and had all the GPS data happily displayed on his instruments, so this reported bug sounds surprising. I had a quick look at the code and the only thing that struck me was the circular buffer of size MESSAGE_STORE_SIZE=16. If not enough space then messages just get dropped. I cannot try to reproduce the problem until my dual display comes back from warranty repair (who knows when that will be). If I do have a problem (my GPS is producing 32 sentences every one second), then I'll try increasing the message store size.

j-lang, 2022-05-04:
This may be right, because when RMC, GGA and VTG is sent by the GPS, as long the GPS has no fix, and RMC sends time and date info, the display shows and updates time and date:
```
$GNVTG,,,,,,,,,N*2E
$GNGGA,054046.00,,,,,0,00,99.99,,,,,,*7B
$GNRMC,054047.00,V,,,,,,,040522,,,N*60
$GNVTG,,,,,,,,,N*2E
$GNGGA,054047.00,,,,,0,00,99.99,,,,,,*7A
$GNRMC,054048.00,V,,,,,,,040522,,,N*6F
$GNVTG,,,,,,,,,N*2E
$GNGGA,054048.00,,,,,0,00,99.99,,,,,,*75
```
With the fix and sending complete position date with GGA, VTG and RMC, the display stop updating time and position data is not shown. 
But only increasing size of MESSAGE_STORE_SIZE and NMEA_SENTENCE-buffer does not change the situation. 
Changed values to: 
#define MESSAGE_STORE_SIZE 250
#define NMEA_SENTENCE_MAX_LENGTH   384 //96 
#define NMEA_SENTENCE_HISTORY_SIZE 96 //24

j-lang, 2022-05-05:
Activate only RMC, VTG and GGA messages (perhaps you made it already)

_1. Rise transmission speed to 115200 Baud for GNSS serial._
Changed speed to 115200 -> no change in shown values on the T110 Display all COG, SOG, lat,lon, time, date show "---"

_2. You can put simple(!) printf's into NmeaDecoder::DecodeSentence to see if you reach decoding stage.
I do not unterstand the the difference between the current NMEA console-output and my added debug output to console: 
Added code
```
	case 0x524D43:
		// RMC sentence
		CONSOLE.print(" DecodeSentence line ");
		CONSOLE.println(__LINE__);
		CONSOLE.print(" RMC-Sentence: ");
		CONSOLE.println(sentenceBuffer[sentenceIndex]);
		DecodeRMCSentence(pField, navData);
		break;
```

Console output (incl. NMEA stream):
```
$GNGGA,101720.00,46xx.x2987,N,01xxx.x3251,E,1,05,2.97,577.7,M,43.4,M,,*45
$INMWV,10.0,R,0.0,N,A*0b
DecodeSentence line 160
RMC-Sentence: $GNRMC,101721.00,A,46xx.x2938,N,01xxx.x3143,E,1.666,,050522,,,A*63
$GNRMC,101721.00,A,46xx.x2938,N,01xxx.x3143,E,1.666,,050522,,,A*63
$GNVTG,,T,,M,1.666,N,3.085,K,A*34
$GNGGA,101721.00,46xx.x2938,N,01xxx.x3143,E,1,05,2.96,577.7,M,43.4,M,,*41
```

j-lang, 2022-05-05:
_You have an T120 Windsensor connected? $INMWV. I have not._
Yes there is a windsensor nearby, that sends $INMWV. 
_Please have in mind that complex print out - like yours - will disturb the timing schedule. Use only few bytes!_ 
-> which timing schedule do you mean? time do decode the nmea sentences and hand over to micronet output routine? 
_OK, the RMC sentences reach this stage. Don't know what you meant with difference? I see no._
... because the NMEA output in this answer: https://github.com/Rodemfr/MicronetToNMEA/issues/4#issuecomment-1114195622 
shows that RMC, VTG and GGA are send to the console -> this output is the result of the nmea-decoding. 

_Now you can try to move forward with print to see where it died._ 
As I understand the GNSS messages are decoded in NmeaDecoder::DecodeSentence and decoded values are stored in navData. 
In MicronetCodec::EncodeDataMessage assembles values from navData into a micronet datapacket. 
COG/SOG Lat/Long  Time and date are valid fields there. 

What I does not understand is, why the time message is inserted two times into the micronet datapacket in EncodeDataMessage?:
```
	if (navData->time.valid)
	{
		offset += Add16bitField(message->data + offset, MICRONET_FIELD_ID_TIME, (navData->time.hour << 8) + navData->time.minute);
	}
```

_And you can look with sdr stick if the transmitted message is changing. And so on._
-> I have a sdr stick, but do you have tips which tools are useful to decode the micronet datapackets?

j-lang, 2022-05-05:
Found a solution by sending alternating time & date and navigation data in EncodeDataMessage. 
Payload length is reduced and now the T112 Display shows time & date, COG/SOG und LAT&LON 
Pull request will follow.

dwarning, 2022-05-05:
Can't believe that a sensor from other network is used.
The entire approach is working in a loop with us resolution. You can calculate how long it needs sending 1byte over console.
I am using recording with URH and rtl-sdr stick. It is shown in the main thread.
Good to hear you find a solution.

j-lang, 2022-05-05:
Maybe there is a timing issue with teensy 4.1 that works with 600 MHz instead of 120 MHz of teensy 3.5. 
The payloadLentgh with lon&lat and cog&sog message is 25, of time&date 13. 
All together have a payload length of 38.

davidwkerr, 2022-05-06:
Got my dual display back a few hours ago.
I confirm that I also have j-lang's problem.
I am about to rebuild and run at 120MHz. Will advise.

davidwkerr, 2022-05-06:
Okay- ran it at 150MHz (closest I can get to 120MHz) and no difference. So I doubt it is anything related to clock speed. The number of sentences from the GPS is clearly a strong influence on it not working properly.

j-lang, 2022-05-06:
> Okay- ran it at 150MHz (closest I can get to 120MHz) and no difference. So I doubt it is anything related to clock speed. The number of sentences from the GPS is clearly a strong influence on it not working properly.

I found a solution, just use code from this push request: https://github.com/Rodemfr/MicronetToNMEA/pull/7 
It seems for me, that the micronet packet is too large the display does not accept it (anyway) -> therefore I send GNSS data in two alternating packets (or long packest are loosing sync - cannot check this in lack of measurement tools)

davidwkerr, 2022-05-07:
Hi, Well done!
I can see the diff files for what you have done and would like to download the files. At least xxxxcodec.cpp, xxxxcodec.h and MicronettoNMEA.ino
But I cannot see how!
Is there a way to get them so I can try before they get merged to Master?

davidwkerr, 2022-05-07:
Okay. I finally got gh cli sorted out and working.
I downloaded your modified MicronetCodec.cpp plus the modified MicronetToNMEA.ino. I don't want the GPS config stuff, which I assume is separate.
Everything builds fine.
However, I still have the problem. No change. Only the magnetic HDG is showing on the displays as well as the hull and wind information. No Lat/Lon, time, SOG (COG not valid stationary).

Incidentally, I am running:
Masthead wind instrument.
"Analogue" Wind Display
Dual digital display.
Hull unit connected to Triducer (speed, temperature, depth).

j-lang, 2022-05-07:
If you do not want GPS initialization, just comment the line
`#define USE_UBLOXM8N`

do have 4 - Start NMEA conversion started (connection via USB) 
GPS Info are only shown if there is a GPS Fix and $GNVTG / $GNGGA have valid content.

davidwkerr, 2022-05-07:
Hi,
Yes, I am well aware of needing menu 4 and having a fix.
None of that was the problem!
The problem was gh (Git cli) and Eclipse "remembering" older files.
So, now it is working!
Thank you!
At the moment, I have reduced the GPS to 4 GPS sentences (GLL/RMC/GGA/VTG) and the 6 $IN sentences and the HDG are all being generated well.

I have the SOG/COG/Lon/Lat/Time on the dual digital display. So, that is fantastic!

The headers of the changed files could do with a short revision history and a few comments on the changed code would be useful. I thought I had the new files but for a while, Eclipse (Sloeber) was sticking with the older unrefreshed files and it was difficult to see what was what.

davidwkerr, 2022-05-08:
I have found my first "Hang" of the software. It was displaying the GPS sentences using option 4 and after several hours, the sentences stopped. Hitting the ESC keys produced the response "ESC key pressed, stopping conversion" then there is no menu displayed and none of the menu items respond to any key. ESC is the only key that produces anything, just the same message each time. So the processor is running but the code is in a loop, only responding to ESC. Of course, all output from MicronetToNMEA to the displays had stopped (which is how I first detected the problem).

davidwkerr, 2022-05-08:
Hm, it has happened again. After about an hour this time. But this time, after several ESCs, I hit "5", then restarted. But it remembered the "4" from previously and had not seen the  "5".

$INMWV,180.0,T,0.0,N,A*35
$INMTW,18.0,C*1d
$INVHW,,T,,M,0.00,N,,K*4c
$GNRMC,090037.00,A,3337.27594,S,15119.18420,E,0.614,,080522,,,A*7B
$GNVTG,,T,,M,0.614,N,1.137,K,A*3A
$GNGGA,090037.00,3337.27594,S,15119.18420,E,1,12,0.97,32.9,M,20.1,M,,*6F
$GNGLL,3337.27594,S,15119.18420,E,090037.00,A,A*6C
$INMWV,255.0,R,0.0,N,A*38
$GNRMC,090038.00,A,3337.27583,S,15119.18430,E,0.290,,080522,,,A*7B
$GNVTG,,T,,M,0.290,N,0.538,K,A*38
$GNGGA,090038.00,3337.27583,S,15119.18430,E,1,12,0.97,32.6,M,20.1,M,,*68
$GNGLL,3337.27583,S,15119.18430,E,090038.00,A,A*64
ESC key pressed, stopping conversion.
ESC key pressed, stopping conversion.
ESC key pressed, stopping conversion.
ESC key pressed, stopping conversion.
ESC key pressed, stopping conversion.

Rodemfr, 2022-05-08:
Hi all,

Sorry for the late answer, I was on vacation far from internet.

I started reading this thread and the troubles you face let me think you have a frequency problem.
Let me explain : to operate properly, CC1101 frequency must be precisely tuned to the one of the Tacktick display. By nature, Crystals are not very precise and need to be "calibrated". When you by a Tacktick device, this has already be done at factory. For MicronetToNMEA, we need to do it. This is the puprose of menu 6 (Calibrate RF Frequency)
This function listens for Display's messages and precisely tunes its frequency to it and stores the calibration factor in flash. However, I recently made changes that can change theses values, making you already flashed configuration erroneous. So the first thing to do is to recalibrate your RF frequency with your display by re-running menu 6. A bad RfFrequency dramatically reduce operating range of MicronetToNMEA.

But that's not all, I recently modified RfDriver.cpp code to supposedly correct the Tx baudrate by the same factor than the RF one. I'm wondering if it is a good fix reading your reports. When the baudrate is not good, small messages sent by MicronetToNMEA are understood by display, but longer ones will be misunderstood as soon a mismatch of one full bit will be reached because of the bad baudrate. This really looks like what you observe whenyou display display heading 
You can try to comment line 43 of RfDriver.cpp and see if things are going better ( float correctedBaudrate_kbaud = 76.8 * (MICRONET_RF_CENTER_FREQUENCY_MHZ + frequencyOffset_mHz) / MICRONET_RF_CENTER_FREQUENCY_MHZ; "

I will check your pull request during the week to integrate them.

Best regards,

Ronan.

davidwkerr, 2022-05-08:
What you say sounds very logical for the base cause of the problem. I had monitored the centre freq. with a high quality spectrum analyser and used menu item 6 (getting offset of about 3.4kHz) and still experienced the problem. I have added a small 916MHz dipole and am getting excellent range (of the Mag HDG) for the CC1101 to Micronet displays. About the same as the Micronet devices. 

I'll see if I can revert the code and do some testing without the scaling of the baud rate. But then, adjusting the CF in the code until the measured offset is just a few hundred Hz should have achieved the same result.

davidwkerr, 2022-05-09:
I've now run over 30 tests of the original master code and with/without the corrected baud rate and different CFs to minimise the offsets. Here is a summary of my observations:

1. One time, I was able to get Lat/Lon/Date/Time/SOG displayed on the Micronet display for about 10secs. With fixed 76.8kBaud and reported offset <1kHz.
2. I really do not think the calculation to baud rate makes any significant difference. That is from my measurements and also the fact that a 5k offset is only a couple of parts per million in the baudrate. It would take a very long packet to be unsynced.
3. If I run multiple option 6 RF calibrations, I get differing offsets every time. Whether or not I do a save. I tried adjusting the CF 50% of the reported offset and recompiling but the reported offsets are still very variable (up to +- 4kHz). So a couple of times I got down to a reported offset of 400Hz and saved that but still no Lat/Lon/Time/Date etc. So the estimation of CF by option 6 seems to have  +-several kHz of variation in it. I really doubt that is caused by fluctuations in the various crystal oscillators.
4. Another thing, here is what I am using now:
Frequency = 915864.56kHz
Range = 71.78kHz
Deviation to real frequency = -0.44kHz 

But what is the "Range=" figure? With very close CFs (+-3kHz), I am seeing figures that are very variable:
Range = 62.81kHz
Range = 74.77kHz
Range = 71.78kHz

I have now put back j-lang's patches and Lat/Lon/Date/Time/HDG/SOG/COG are again displayed. However the fact that I got it to work for 10secs one time with the old code does indeed suggest that there is something that could be improved in the syncing of longer packets.

davidwkerr, 2022-05-09:
I have measured my Micronet CF (near Dual Display- master) three ways. Here are the answers:

1. 915.914MHz. High quality Spectrum Analyser with crystal oven. Averaged 10mins.
2. 915.913MHz RTL SDR BLOG shielded SDR.
3. 915.861MHz with 1kHz offset. MicronetToNmea freq calibration
I do not understand #3, which is 52kHz out (57ppm!)

Rodemfr, 2022-05-09:
You have 916MHz tacktick devices. So I understand that you are in the US.
MicronetToNMEA probably has some implicit references to 868MHz frequency. Let's tackle them.
I did not have time to look at j-lang's patchs. I will do this this evening.

From you latest messages, I understand the following :
- RF center frequency found by MicronetToNMEA is good by a few kHz. The frequency finding algorithm seems working well.
- You still have display issues which really look like a baudrate problem.

For the second point, I suspect the following : all constants used by MicronetToNMEA are "guessed" by reverse engineering. I use poor quality tools to find them (uncalibrated RTL-SDR). It might very well be that the baudrate value used by MicronetToNMEA is not the correct one. Close to the ideal one but far enough to cause troubles.

So I think we first have to search for the right baudrate. Currently it is hardcoded at line 43 of RfDriver.cpp with a value of 76.8kbaud.

> 915.861MHz with 1kHz offset. MicronetToNmea freq calibration
> I do not understand 3., which is 52kHz out (57ppm!)

I think the cause of the deviation is the same than for the baudrate : MicronetToNMEA guesses the absolute frequency by trusting a frequency constant which is likely wrong.

davidwkerr, 2022-05-09:
Very useful info. Thank you for all the work you have put into this. Thanks also for the very clear explanations.
I am in Australia.
When I get a chance, I will soon do some work looking at the baud rate.

davidwkerr, 2022-05-09:
I just ran a simple experiment increasing and decreasing the baud rate by 50baud each time to find the upper and lower limit for successfully decoding Lat/Lon etc with j-lang's patches (so as to get shorter packets). Lowest working I got was 76.50kb and highest 77.25 giving a centre of 76.875kbaud. The difference between 76.8 and 76.875 is 0.012usecs per bit so 500bits would be the max packet length to move half a bit time (if there is no timing correction in the demodulator). I'll see if I can refine it a bit more...

Rodemfr, 2022-05-09:
This more or less means that we must reduce the size of messages sent by MicronetToNMEA to the display. I started a change which will make the amount of data fields sent by MicronetCodec.cpp easily configurable. Hopefully, this will be ready this week-end.

davidwkerr, 2022-05-09:
Unfortunately my RTL-SDR-BLOG does not work with my URH otherwise I could do more scientific measurements for the baud rate. I do have an 8 digit freq counter with oven which has been calibrated against an atomic clock 2 yrs ago. Will try and measure the xtal on my CC1101 (26MHz) but this might be too difficult unless it has a buffered output. The CF I get from my accurate spectrum analyser agrees with that from the RTL but the reported CF from the MicronetToNMEA is a bit different as can be seen in my earlier post. I shall go back to the unpatched software with different baud rate and see how that goes.

davidwkerr, 2022-05-09:
From the TI datasheet. The Micronet display serial/part numbers have different numbers for the default 865 and default 916MHz systems. 

"For compliance with modulation bandwidth requirements under EN 300 220 in the 863 to 870 MHz frequency range it is recommended to use a 26 MHz crystal for frequencies below 869 MHz and a 27 MHz crystal for frequencies above 869 MHz."

davidwkerr, 2022-05-10:
If Raymarine does use different xtals in the units for the two different systems, then these are the baud rates:

26MHz           27MHz
76.965           77.0416
76.767           76.8356
76.568           76.6296

So, if the 865 system is running 26MHz and the 916 is 27MHz, the bit times are 13.0264uS and 13.0148uS respectively. So, the half bit time (6.5uS) would be reached after approximately 70bytes. That is probably a best case figure because there are crystal temperatures and tolerances that also come into play.

So, it is only a theory, but might explain why I cannot get the bigger packets to decode. But I think j-lang has an 865 system so it would not explain him seeing the same thing. So maybe my theory/guess is wrong. I have tried to pick up the crystal oscillator with a sensitive Network Analyzer placed next to the Micronet instrument but cannot detect it. Which is probably good from a RFI perspective but does not advance the information any further.

Rodemfr, 2022-05-10:
I think you found the root cause.

I see in CC1101's datasheet that several calculations depend on xtal frequency, including baudrate. However, arduino's CC1101 driver uses many hardcoded constants which are probably bound to 27MHz xtal. For center frequency, the error is compensated by the frequency calibration algorithm which simply shift it to the same one of the display. But for baudrate, there is no such compensation.

I can imagine such a calibration algorithm for baudrate : We could artificially add 10 or 20 fake devices into the network to force master display to send a very long 0x01 "master request" message. This message could only be decoded by MicronetToNMEA with a good baudrate. But I'm not sure this will practically work but that's worth to try.

Also, I fear that other parameters than baudrate could be wrong, degrading the quality of reception.

Rodemfr, 2022-05-10:
I just created a new branch for investigation : investigate_baudrate_issues
It now reduce the size of each message by cycling around 4 configurations :
- TIME + SOG/COG + HDG
- DATE + POSITION + HDG
- XTE + DTW + HDG
- BTW + VMGWP + HDG
Now I will investigate a way to autodetect the best baudrate

Rodemfr, 2022-05-10:
I used exactly the algorithm of frequency calibraton to calibrate baudrate : this works fine ! I weep from 76800 - 1500 to 76800 + 1500 by steps of 50 baud and the algorithm found the correct baudrate on my setup (76875).
I need to clean-up a bit the code before pushing to Github but this should be ready tomorrow.

davidwkerr, 2022-05-11:
I found the driver is not setting FOFFG so I added:
SpiWriteReg(CC1101_FOCCFG,      0x1F); 
to the if and the else of "void ELECHOUSE_CC1101::setCCMode(bool s)" in the ELECHOUSE driver cpp code.
It made an improvement of 2dB in my received signal. So worth doing.

The above sets 4* pre sync word gain and 50% after detection. It also moves the baud rate +-50% of the channel BW. I also tried finer variations (12% for 0x1D and 25% for 0x1E) and could not tell the difference. As the register was not set by the driver, I think that disabled the compensation mechanism entirely. So I think this is worth doing to slightly improve reception.

Rodemfr, 2022-05-11:
I pushed the new baudrate calibration code. For now, the sweep range is hardcoded. If you need to extend it for you 916MHz configuration you can do it at line 1013 of MicronetToNMEA.ino
Be careful that I changed the configuration structure stored in EPROM to add this new parameter. It means that you will have to reconfigure your device, especially RF center frequency. RF celibration should be run before baudrate calibration of course.
For the message split, the biggest message should be 38 bytes.
Let me know if it solves your issues.

davidwkerr, 2022-05-11:
Thank you!

The short answer is that it all works very well. No problems.

The second part of the answer is relatively minor. When I did the RF centre freq setting, I thought I would also try your new baudrate scanning code. That gave an offset of 500baud which I saved. But then I was stuck! Doing it again and saying yes would again put a poor value into EEPROM and saying no would leave it unchanged. So I had to change the code just to get back to 76.80kbaud, which worked fine. So there needs to be a way to reset back to "normal" ("normal"=76.8 is good for my 916 and presumably also for your 868).

So, I recommend the following simple addition around line 1111 of MicronetToNMEA.ino

	else
					{
						gConfiguration.baudrateOffset_baud = 0;     //Reset the offset
						gConfiguration.SaveToEeprom();                  //and save to EEPROM 
						CONSOLE.println("Configuration discarded & offset zeroed");

As I mentioned, the algorithm found a value that was too high. Doing a second time it got worse. Here is a scan I did after resetting the offset. You can see the extra * hits on the right hand side, moving the average too high. I'm not sure why this might be happening:

.................................***********************.*........********.......
Baudrate = 77450.00 baud
Range = 2000.00 baud
Deviation to real baudrate = 650.00 baud

davidwkerr, 2022-05-12:
Ah, a real (but small) bug. RF output test has become menu item 10 but you cannot enter two digits.

Rodemfr, 2022-05-12:
Thanks for the feedback.
This very strange "matching area" of the baudrate sweep algorithm on the right of your snapshot really let me think that there is something wrong with SmartRF-CC1101 driver when using 915MHz range.
So here is the plan : I will not include this baudrate search algorithm in master branch. We are not supposed to search for the baudrate it should be exactly 76800 baud. Only CF needs to be calibrated. I will include the message split in the master to allow 915MHz configurations to work properly with the current CC1101 driver. I will also include j-lang's code to initialize M8N GPS. Once all of this will be done, I think I have no other choice than to rewrite the CC1101 driver.

davidwkerr, 2022-05-12:
There is a PM for you (Rodemfr) on the YBW Forum. I agree re baudrate. We know what it is and just setting it to be fixed should be fine, even with the worst of 26MHz crystals. Implementation of FOFFG means there is some latitude and "fine tuning". 

The CF calibration is a good thing to have.

My high quality 26MHz and 27MHz crystals arrive on Monday and I might get the chance to see how 27MHz and 916MHz go together. It depends how steady my hands are with the SMD rework. I have a spare CC1101 board which I can afford to wreck.

Rodemfr, 2022-05-13:
I updated the baudrate branch :
- Remove baudrate search algo
- Added M8N initialization code
- Added a way to select 868/915 frequencies in BoardConfig.h

I used 915.914 MHz as reference CF for US/AU, as stated in your previous message.
Configuring MicronetToNMEA for non-EU freqs should be easy now, just change FREQUENCY_SYSTEM define in BoardConfig.h.
Setting GNSS_UBLOXM8N to 1 in BoardConfig.h will enable M8N initialization code (thanks j-lang).

If you confirm me there is no issue with these fixes, I start rewritting CC1101 driver.

davidwkerr, 2022-05-14:
Hi Ronan,
I confirm it is all looking okay. However I did not test the M8N code (my hardware Rx is not connected).
I double checked the CF and confirm your figure is within 5kHz of what I am measuring on my spectrum analyser. My crystal is somewhat off, so my reported number is about 60ppm low. That might change when I replace the crystal. But what you have is good.

Because of the uncertainty re DEVIATN register, I have hardcoded my copy to set 0x44.

Per the discussion with dwarning, I added (916MHz) clb settings to my copy:

<<cc1101Driver.setClb(4, 77, 79);  //Set the clb values derived from calibration script>>

I put this at the end of RfDriver::Init in RfDriver.cpp.

When you are rewriting (simplifying!) the driver  the FSCAL registers from RFStudio might be too generic because they do not change with CF. So perhaps start with the above clb figures for 916 and dwarning quoted 3, 17, 19 which I assume are for 868. It is fairly easy to turn these into FSCALx register values from the current driver code or to leave the clb code "as is" because the calibration script does it that way (reports clb figures not FSCAL values).

For now, in my copy of the code, I am directly writing the DEVIATN register with 0x44 rather than setting 34kHz to get 38kHz because I think the current driver code is not necessary and just confusing. I think 0x44 is a good value which is independent of 916/876 and matches (on my spectrum analyser) very, very closely to (certainly my own) Micronet devices.

Overall, the system is looking great with good range and everything being decoded well. It is still in my house (going through walls) but in about 10days the wind instrument and anenometer are going onto my boat. The hull transducer will be installed some months in the future when annual slipping happens.

I am working on my box with PCB to interconnect Teensy 4.0, magnetometer, CC1101 board with ext antenna, switching regulator, differential NMEA in/out, UDP packet transmission, Ublox M8N (Drotek) module. I will try out a 27MHz crystal some time next week, when they arrive from the USA. I might change my 26MHz for a better quality one because it might provide better long term stability.

Best Regards,
David

Rodemfr, 2022-05-14:
Thanks for the feedback !

davidwkerr, 2022-05-14:
I checked the calibration of my spectrum analyser and RTL-SDR-BLOG today and they are each better than 1ppm.
So I checked the deviation from the CC1101 very accurately. I measured 76.25kHz actual versus 76.172kHz predicted by RfStudio with DEVIATN of 0x44. That's good! 
My 26MHz crystal is 80ppm low, which is a lot less good. But those boards were only 1.5Euros each so I am lucky the crystals even oscillate!

dwarning, 2022-05-14:
David,
my test script has a random sequence. It the reason for the broad spectrum.
If I look with SDR++ and airspy to the spectrum with different DEVIATN settings (independent direct write the register or using the setDeviation() function) I see that spectrum changes slightly but not really the deviation. How are your observations?
Perhaps we overlook something.

davidwkerr, 2022-05-14:
I found the deviation did change. But with these various other changes, and because I have now done equipment checks and more precise measurements, I shall try again perhaps with 0x42 which should give a total of 64kHz. Will let you know very soon!

davidwkerr, 2022-05-14:
Okay, I have done it and I observed the same as you. The modulation looks different and when I average the waterfall, the max and min deviation is much more spread. The average is almost the same as with 0x44 (just a few hundred Hz different). The range is still good and I cannot tell the difference froma performance perspective. But I would say that the deviation is basically unchanged but very spread! I am about to try 0x47. 
Okay, that is even worse. More spread spectrum though same average deviation.
I went back to 0x44 and it is definitely the best looking spectrum with "crisp" deviation.

davidwkerr, 2022-05-14:
Here is the equation. 
[image]
where DEVIATION_M=0->7 and DEVIATION_E=4. So:
26*10exp6/2exp17(8+0to7)2exp4

davidwkerr, 2022-05-14:
Which should gives a total deviation range (DEVIATN of 0x40 to 0x47) of 50.78kHz to 95.21kHz. 0x44 gives 76.17. But the reality looks different on the spectrum!

dwarning, 2022-05-14:
In the end spectrum and center frequency should similar like display has transmitted. Compare can be reached by moving one of the transmitter near to your test receiver.

davidwkerr, 2022-05-14:
Sorry, I do not understand what you are saying.

davidwkerr, 2022-05-18:
FYI, I swapped the xtal on one of my CC1101 boards for a 10ppm high quality crystal and it is working well. Much closer to 26MHz than the cheap one. It would be even closer if I changed the load capacitors but they are too small for me (0402). Also, this current branch of the software has been running well now for over 4 days.

I realised there is little point in trying a 27MHz crystal with the CC1101 because I think the Micronett devices still use a CC1000 chip which usually has a 14.xxMHz crystal. Certainly nowhere near 26/27MHz. So it is probably not possible to get an exact baudrate match, hence the strategy of limiting packet length is good.

I am going to try a little experiment reading the FREQEST register and putting it into the FSCTRL0 register to pull the CFs together. If it works, it probably will not make any difference because the tuning option gets them within 3kHz (which should be perfectly good enough) but it will be interesting. The CC1000 has no such feature and from my observations, the Micronet devices do not seem to do any CF adjustment,

Rodemfr, 2022-07-11:
The latest version of master branch solves this. It creates four "virtual" slave devices so that each of them will only transmit small messages, avoid the "unprecise baudrate issue".

closed 2022-07-11 by Rodemfr