Rodemfr/MicronetToNMEA #11: No Micronet signal receiving with last version

j-lang, 2022-06-21:
I have my prototype running with firmware with code of my last pull request of 13th of May. 
Today I downloaded, configured uploaded the code (commit: [9bae17b](https://github.com/Rodemfr/MicronetToNMEA/commit/9bae17ba3172276f3fedee532d17e835d96d54f7))
The first test shows:
Initializing CC1101 ... OK
Initializing navigation compass ... LSM303DLHC Found

But 2 - Scan Micronet networks shows:
/!\ No Micronet network found /!\
Check that your Micronet network is powered on.

If the "old" firmware is uploaded into the board, the Micronet communication starts immediately. 

The new parameter is set correct! : 
// 0 -> EU (868Mhz)
// 1 -> non-EU (915Mhz)
#define FREQUENCY_SYSTEM 0

Hardware:
Teensy 4.0

j-lang, 2022-06-21:
Yes I tried, but there are only dots, no asterisk for received data as in the old version.

Rodemfr, 2022-06-21:
Damn, it works for dwarning and me but not for you... not good ! I understand that you also are running at 868MHz, right ?
The only thing I can do is a careful review of the modification and try to find what could explain this.

dwarning, 2022-06-21:
So try to change the recommended line.
If I compare your picture for 4.0 with mine 3.6 the default SPI pins are same. Mayit is only a KiCAD problem with the names.

Rodemfr, 2022-06-21:
Default config for GDO0 is pin 24 which does not exists on Teensy 4.0. Maybe this modification has been lost during your last pull ?

j-lang, 2022-06-21:
I double checked configuration of old and new version and they should be the same:
Old:

```

// Selects on which I2C bus is connected compass as per Wiring library definition
#define NAVCOMPASS_I2C Wire

// CC1101/SPI pins
#define CS0_PIN  10
#define MOSI_PIN 11
#define MISO_PIN 12
#define SCK_PIN  13
#define GDO0_PIN 14
```

New:

```
// Select EU (868MHz) or not-EU (915MHz) Micronet frequency
// 0 -> EU (868Mhz)
// 1 -> non-EU (915Mhz)
#define FREQUENCY_SYSTEM 0

// Selects on which I2C bus is connected compass as per Wiring library definition
#define NAVCOMPASS_I2C Wire

// CC1101/SPI pins
#define CS0_PIN  10
#define MOSI_PIN 11
#define MISO_PIN 12
#define SCK_PIN  13
#define GDO0_PIN 14
```

dwarning, 2022-06-21:
Interesting, whenever complicate discussion [https://forum.pjrc.com/threads/62340-SPI-transfer()-Hangs-on-in-the-first-call-using-Teensy-4-0-and-CC1101](url) using teensy 4.0 and cc1101. They claim not to use 
`	pinMode(SCK_PIN, OUTPUT);
	pinMode(MOSI_PIN, OUTPUT);
	pinMode(MISO_PIN, INPUT);
	pinMode(CS0_PIN, OUTPUT);
`
in init section. This seems teensy 4.0 specific!
And teensy guru Paul Stoffregen prefers a 1ms delay after set SS and SPI transfer.

j-lang, 2022-06-21:
@dwarning; as I understand the additional delay is necessary starting in  CC1101Driver.cpp line 53: pinMode(SCK_PIN, OUTPUT); ....
Is it correct that they understand SS as CS0? 

There is a similar post https://forum.pjrc.com/threads/61932-Different-SPI-Behavior-on-Teensy-4-0-and-4-1] with the assumption that this SPI issue will not be in Teensy 4.1 

I tested this with Teensy 4.1 board, but there is the same issue: CC1101 is initialized but does not receive Micronet data. 
On Teensy 4.1 the original GD0 pin is used:

```
// CC1101/SPI pins
#define CS0_PIN 10
#define MOSI_PIN 11
#define MISO_PIN 12
#define SCK_PIN 13
#if defined(ARDUINO_TEENSY40)
    #define GDO0_PIN 14
#endif
#if defined(ARDUINO_TEENSY41)
    #define GDO0_PIN 24
#endif
```

dwarning, 2022-06-22:
A bit off topic - I just seeing that my link above didn't work. I used "Add a link" but it by clicking on it a github issue page is opened. Same behavior on you?
https://forum.pjrc.com/threads/62340-SPI-transfer()-Hangs-on-in-the-first-call-using-Teensy-4-0-and-CC1101
@j-lang: Say said delay(1) should only used before first SPI transfer action. Can't say where the place is in the code, perhaps just after beginTransaction line 65.
Yes, SS (slave select) and CS (chip select) are same meaning in this context.

dwarning, 2022-06-22:
BTW: You can access pin24 on 4.0 board back side (the stripe left above).

Rodemfr, 2022-06-22:
I just pushed a change in SPI driver in "rewrite_cc1101_branch". I now use SPI as advised in Arduino's doc. This *might* solve your issue.

j-lang, 2022-06-22:
Replaced CC1101Driver.cpp and CC1101Driver.h from your branch, but there is no change. 
I checked the "ISR-problem" assumption and added a console output into void RfDriver::GDO0RxCallback(). 
"Old" firmware gives permanently an output but the new version seem to have no IRQ on received data.

Rodemfr, 2022-06-22:
I still suspect SPI problem. I got a similar behavior when I rewrote SPI functions. The first command is understood by CC1101 but not the following ones. Since the first command is just a read to check IC presence, you get detection but all after is not working, leading to a non reception of packets. Teensy 4.x is a lot faster than 3.5 so this is likely a speed problem. Looking to page 30 of CC1101 datasheet I see there are a lot of timings to respect. Data setup time is around 76ns and can potentially bit hit by a 600MHz CPU. I will add a small delay between each byte in burst reads.

dwarning, 2022-06-22:
Checked it too with my 3.6 - not any problem!
Opposite - by transferring the entire buffer in the both Burst routines we have real burst write and read.
`void CC1101Driver::SpiWriteBurstReg(byte addr, byte *buffer, byte num)
{
	byte temp;

	temp = addr | WRITE_BURST;
	digitalWrite(CS0_PIN, LOW);
	SPI.beginTransaction(spiSettings);
	while (digitalRead(MISO_PIN))
		;
	SPI.transfer(temp);
        SPI.transfer((uint8_t *) buffer, (uint32_t) num);
	SPI.endTransaction();
	digitalWrite(CS0_PIN, HIGH);
}
void CC1101Driver::SpiReadBurstReg(byte addr, byte *buffer, byte num)
{
	byte temp;

	temp = addr | READ_BURST;
	digitalWrite(CS0_PIN, LOW);
	SPI.beginTransaction(spiSettings);
	while (digitalRead(MISO_PIN))
		;
	SPI.transfer(temp);
        SPI.transfer((uint8_t *) buffer, (uint32_t) num);
	SPI.endTransaction();
	digitalWrite(CS0_PIN, HIGH);
}`
Perhaps we should consider this improvement later after solving the 4.0 timing problem.

dwarning, 2022-06-22:
> You are right: with 24MHz the module receives data, with 150MHz and above, not. (tested on Teensy 4.0 and 4.1)

Which frequencies you talking about? You changing cpu clock? Why not changing SPI clock in settings?

j-lang, 2022-06-22:
Changing of SPI clock had no effect.

Rodemfr, 2022-06-23:
I added small delays in SPI transfers where requested by datasheet. You can test in with the latest version of "rewrite_cc1101_driver" branch.

Rodemfr, 2022-06-24:
> Checked it too with my 3.6 - not any problem! Opposite - by transferring the entire buffer in the both Burst routines we have real burst write and read. `void CC1101Driver::SpiWriteBurstReg(byte [...]

This modification is cleaner/faster but might produce too fast transfers for CC1101. According to datasheet, we have to ensure at least 76ns of delay between two bytes in burst mode. With a fast CPU like Teensy 4.0, we will likely be too fast. I'm waiting for j-lang's results of my latest modifications but I strongly suspect this is the reason why it is not working on his board.

dwarning, 2022-06-24:
I can confirm latest change work for me too with my 3.6. I am using buffer transfer for both burst routines, but with same 100ns delays as in the non burst functions.
One question: Why we can not reach the 76ns between bytes with clock setting or delay inserting.
Only polishing: line 467, you can use macro name for DEVIATN register instead 21.
I reduced PA power to 5dBm (0x86) to save current. Works for me more than 10m with to walls between.
BTW, for higher levels the PATABLE is frequency dependent. We need two tables.

j-lang, 2022-06-24:
Unfortuanately there is no change: it works on Teensy 4.0 with 24MHz CPU-Clock, but with higher rates not. 
CC1101 driver from: https://github.com/Rodemfr/MicronetToNMEA/commit/d0c4ddb886ec5397948b0d2824025eaed6fa46ae

Rodemfr, 2022-06-25:
@j-lang , have you a digital scope to watch how are actually shaped SPI signals ? If you haven't I will have to order a Teensy 4.0 to debug this...

j-lang, 2022-06-26:
> @j-lang , have you a digital scope to watch how are actually shaped SPI signals ? If you haven't I will have to order a Teensy 4.0 to debug this...

I try to get one from a friend within the next days, it has two channels. 
Please define which signals should be recorded.

Rodemfr, 2022-06-26:
SCLK and CS/SS would be perfect. It will allow me to compare your timings with mine.

j-lang, 2022-06-28:
Measurement @24MHz (working): SCLK (red) CS (blue)
[image]

Measurement @600MHz (not working): SCLK (red) CS (blue)
[image]

Unfortunately I got only one probe, therefore the CS-signal looks very bad.

Measurement @24MHz (working): MISO(red) CS (blue)
[image]

Measurement @600MHz (not working): MISO (red) CS (blue)
[image]

Measurement @24MHz (working): MOSI(red) CS (blue)
[image]

Measurement @600MHz (not working): MOSI (red) CS (blue)
[image]

Measurement @24MHz (working): GD0(red) CS (blue) (triggered on GD0, only this diagramm)
[image]

Measurement @600MHz (not working): GD0 (red) CS (blue)
[image]

dwarning, 2022-06-28:
First comment: The bundle of 8 clock periods needs in both cpu speeds same time: ~2us. Means spi clock has around 4MHz and seems independent from this large cpu clock deviation. 4MHz is normal speed.
GDO0 seems to work.
I don't know Picoscope. Which Series is it? Can you catch longer duration and more points? What means the 1ks entry box in the menu? Is it the sample rate? Can you rise this?

Rodemfr, 2022-06-28:
Thanks j-lang, this should be enough to solve the issue.
When I compare your records with those of dwarning, I notice that CS is staying far less than 1us to high state between two acesses in your case while it is about 1us ifor Teensy 3.5. I will add extra delay here. A new version will be ready this evening...

dwarning, 2022-06-28:
@Rodemfr :You are faster - as usual.
@j-lang: Please rise the sample rate to more then 1kS.

Rodemfr, 2022-06-28:
I pushed a potential fix in "rewrite_cc1101 driver" branch. I increased existing delays from 100ns to 1us and I added a 1us delay between CS0 High and LOW. This should to the trick !

dwarning, 2022-06-29:
Now the new cc1101driver:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Rodemfr, 2022-06-29:
Interesting ! We see that the newer driver is slightly slower, which is normal since I added 1us in many places.
Also the newer driver is not reconfiguring the pins at each transfer, avoiding the slow discharge of MOSI pin between transfers.

j-lang, 2022-06-29:
> I pushed a potential fix in "rewrite_cc1101_driver" branch. I increased existing delays from 100ns to 1us and I added a 1us delay between CS0 High and LOW. This should do the trick !

Am I right if I have only to take these files: CC1101Driver.cpp, CC1101Driver.h from [rewrite_cc1101_driver] (https://github.com/Rodemfr/MicronetToNMEA/blob/rewrite_cc1101_driver/CC1101Driver.h) branch? 
Because I receive nothing neither at 24MHz nor on 600MHz: 
```
Scanning Micronet networks for 5 seconds ... done

/!\ No Micronet network found /!\
Check that your Micronet network is powered on.
```
If i jump back to previous version, it works on 24 MHz. 

And another behaviour: 
at 600 MHz the software hangs up at boot (I have only a Teensy 4.1 board with CC1101 connected - the other board is installed in the boat) 
```
Initializing CC1101 ... OK
Initializing navigation compass ... NOT DETECTED
```
Sometimes in other firmware versions, the Menu was not sent to the serial console, but after pressing the ESC key it appears, with latest software running at 600MHz there is no futher action possible.

dwarning, 2022-06-29:
@j-lang: In case you don't know git:
git checkout rewrite_cc1101_driver
git pull
git status
I then copy the latest changed files to my arduino working dir. Let BoardConfig.h as it is, because I have few changes.
Hope it helps.

Rodemfr, 2022-06-29:
Using CC1101Driver.cpp and .h should work.
You have a strange behavior. I think I have no other choice than ordering a Teensy 4.0. This will take some time before I'm able to debug here...

dwarning, 2022-06-29:
@j-lang : Why not using my simple test scripts for receiving and transmitting which I published here and in other Micronet forum places.
The Converter application is very complex and needs a step by step working flow with these simplified checks by using identically board config.

dwarning, 2022-06-29:
Got just now my 4.1. Must solder header. Now started investigation.
First facts: SCK can't reassign to pin 14 - it must 13. Default clock is 4MHz.
My simple receive script works, but the scope diagram is stupid.
Must see further.

Rodemfr, 2022-06-29:
Good that you have a Teensy 4, I will delay my order...

j-lang, 2022-06-29:
> @j-lang : Why not using my simple test scripts for receiving and transmitting which I published here and in other Micronet forum places. The Converter application is very complex and needs a step by step working flow with these simplified checks by using identically board config.

will do it as soon as possible, have very little time.

dwarning, 2022-06-29:
I am in doubt that I am able to solve it.
Separate GNSS and compass is working.
But I have same effect as @j-lang: program hangs. Means setup is not starting PrintMenu. The RestartReception before is in a loop because it seems the board receiving and RfIsr runs.
Sorry.

Rodemfr, 2022-06-30:
This hang is strange. The only code I added is a call to micros(). I will attempt some modifications tonight to try to solve this. I suspect that some of the calls inside SPI functions are forbidden in ISR...

dwarning, 2022-06-30:
I believe the changes in RestartReception and/or the modified SPI init (not anymore in the setup) has to do with the problem. I put a serial.print into the function and it will never end - independent if display is on or not. So the program hangs in this function called in the setup and comes not further.

dwarning, 2022-06-30:
These are the SPI pattern, which come periodically never ending. Perhaps you can assign them to the functions in the RestartReception function:

[image]
[image]
[image]
[image]
.

Rodemfr, 2022-06-30:
By the way, its on branch "rewrite_cc1101_driver"

dwarning, 2022-06-30:
Yes, but header was wrong or not actual. I repaired it.
Unfortunately the hang is still there in setup, putty:

Initializing CC1101 ... OK
Initializing navigation compass ... LSM303DLHC Found

Could you inspect my pictures above? Any information from that? You need explanation?

dwarning, 2022-07-04:
@Rodemfr: You are still engaged with the hang problem?
I am exclude 4.1 SPI and CPU speed as root cause. I compared most settings between old and new driver approach. Most of them are same and plausible. Onlyest I can't follow is the handling of GDO0 settings with the fifo check and CC1101_IOCFG0 setting.
Anyway I don't suffer problems with the old driver lib neither with stability nor with message length. Or better - I didn't found a way to check it.
Meanwhile I connected Serial8 to NodeMCU serial board with a NMEA bridge sketch. Can receive it over WiFi in openCPN.

Rodemfr, 2022-07-04:
Hi dwarning,
I was not home for the week-end. I agree with you, SPI does not explain everything. I now suspect a concurent access issue between IRQ and normal context with SPI driver, or something like that. I need a Teensy 4 to progress on the topic. I just ordered one.

dwarning, 2022-07-07:
Hi Ronan,
I want only inform you that your branch micronet_slave_class show no hang anymore on teensy 4.1. I am using pin 13 for GDO0.
Console show continuous stream but display receives nothing. Check with stick show only display is sending. SPI signal is attached. It looks cc want response immediately to see in 3rd picture.

[image]
[image]
[image]

Here is GDO0 instead of MISO.

[image]

Rodemfr, 2022-07-07:
Thanks,
I just received my Teensy 4.0. I hope to be able to debug this this week-end.

dwarning, 2022-07-07:
Sounds fine.
BTW: I am using pin 13 for SCK. Was wrong above.

dwarning, 2022-07-08:
@Rodemfr:
Perhaps it is the right point in time: Can you take a look into the attached BoardConfig.h. I am open for discussion and modifications.
[BoardConfig.h.txt](https://github.com/Rodemfr/MicronetToNMEA/files/9073173/BoardConfig.h.txt)

Rodemfr, 2022-07-09:
Your BoardConfig looks likes the one I made for my Teensy 4.0 : sounds good.

I connected my Teensy 4.0 to a CC1101 with an older version of MicronetToNMEA and I observed the same than j-lang : no communication at all. I works at 24MHz but not at 600MHz.

I had to add really long delays between each byte in SPI transactions (several microseconds) else I read corrupted data from RX FIFO. Now It seems to read correct data from CC1101 in GDO0 ISR but then GDO0 ISR is not triggered a second time to get the rest of the message from CC's FIFO. I'm unsure of the reason, but I suspect that GDO0 is not going high because of the delays I added in SPI transactions. To make GDO0 return to high state, I need to empty the FIFO which may not be possible if I read too slowly compared to the 76800 baud input flow.
Continuing investigations...

dwarning, 2022-07-09:
Perhaps not the problem, but few warnings (Sorry the code insert format is not working):

In file included from /home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetSlaveDevice.cpp:31:0:
/home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetSlaveDevice.h: In constructor 'MicronetSlaveDevice::MicronetSlaveDevice()':
MicronetSlaveDevice.h:67: warning: 'MicronetSlaveDevice::dataFields' will be initialized after 
  uint32_t dataFields;
           ^
MicronetSlaveDevice.h:66: warning: 'uint32_t MicronetSlaveDevice::networkId' 
  uint32_t networkId;
           ^
MicronetSlaveDevice.cpp:54: warning: when initialized here 
 MicronetSlaveDevice::MicronetSlaveDevice() :
 ^
In file included from /home/dietmar/Projects/Arduino/MicronetToNMEA/RfDriver.cpp:8:0:
/home/dietmar/Projects/Arduino/MicronetToNMEA/RfDriver.h: In constructor 'RfDriver::RfDriver()':
RfDriver.h:49: warning: 'RfDriver::frequencyOffset_mHz' will be initialized after 
  float frequencyOffset_mHz;
        ^
RfDriver.h:47: warning: 'int RfDriver::nextTransmitIndex' 
  int nextTransmitIndex;
      ^
RfDriver.cpp:20: warning: when initialized here 
 RfDriver::RfDriver() :
 ^
***********************************
Regarding BoardConfig.h: You are using pin13 for SCK? The builtin LED is blinking for me, because connected. 

One difference to teensy 3.x boards: pin24 - our GDO0 pin - has no other function in 3.x, for 4.x boards it has amongst others I2C! Do you tried ever another pin?

At last: Can I scope special cases?

dwarning, 2022-07-10:
Made an compare of registers between working and non working teensy 4.1 version. 
Only PKTCTRL0 and 1 are different:
	working 	not working
IOCFG2    B		8
IOCFG1    2E	3
...
PKTCTRL1  84	80 <- no status (RSSI and LQI) appended
PKTCTRL0  4		44 <- data whitening on

IOCFGx will not have impact because GDO1 and 2 are not used.

Rodemfr, 2022-07-10:
I think the Teensy 4.0 issue is fixed. The root cause of the problem is that CC1101 needs ~8us delay between two consecutives SPI commands (i.e. between CS high and CS low), likely to process it. This additionnal delay coupled to Teensy 4 CPU very high speed revealed weaknesses in GDO0 ISR handling and also an error in timing calculations.
I fixed all of this and it seems to work well now with both Teensy 3.5 & 4.0.
Because of an error of mine, the fix is on branch "micronet_slave_class" instead of "rewrite_cc1101_driver".
@dwarning & @j-lang : can you test this and tell me if it's OK ?.
If yes, I will merge the work in master branch and clean-up a bit the mess in the branches.

This error of branch adds a new feature : MicronetToNMEA now presents itself as 4 independent slaves devices, allowing a 1s refresh for all data while avoiding baudrates problems with large messages.

j-lang, 2022-07-10:
Great to read this, I will test this within the next days.

dwarning, 2022-07-10:
OK, its better now but not complete:
With this longer CS phase receiving on SPI is working, see picture and compare to my former post:

[image]

I can go down to 6us for DELAY_BETWEEN_CS_US - with 5us I got garbage. I will stay with 8us.
Sending is better too, we have separate messages from display and converter. But in the end the display shows no values. Have to investigate more the messages. 

You go to pin9 with GDO0 for T4.1. I am really insecure about the reconfigure capabilities of the series 4 teensys! 9 is an output you reconfigure to an input. Perhaps pin 5 would be better. You found an problem with pin 24? I left it in my config.

And can you please answer my 2 findings of register deviation in my former post. Do yo want data whitening and do you want omit status bytes?

Rodemfr, 2022-07-10:
Concerning the register deviation you found, these non working values are not supposed to be written. So I suppose the writings were corrupted because of timing, like the type of corruption I have seen whenreceiving FIFO data.
I did not use pin 24 because it is not easily reachable on Teensy 4.0 : this is a pad below the board, not a hole. I used pin 9 because it is next to CS, so my soldering are in the same area. I guess any digital pin can be used for this.
I did not test Teensy 4.0 w.r.t sending data to Display, only receiving. I only tested sending on Teensy 3.5.
I will do this next week.

Rodemfr, 2022-07-10:
I confirm that data sending is not working as well on Teensy 4.0 than on Teensy 3.5. I will investigate this later, likely not before end of next week.

Rodemfr, 2022-07-10:
@j-lang, wait before testing. Tx still has issues with Teensy 4, I need to investigate further...

Rodemfr, 2022-07-10:
Sorry, I made a mistake, Teensy Tx is working well. Bad manipulation...
@j-lang : you can test it.

j-lang, 2022-07-10:
Yes can receive time and date from GPS immediately. Running Teensy 4.1 with 600MHz and GD0 on Pin 24. 
It also works on 24 MHz. I prefer the 24 MHz config because it takes ca. 60 mA less than with 600 MHz (value from cheap USB tester) 

Currently I have no compass on my lab-config. 
I use Platform.io. 

```
#if defined(ARDUINO_TEENSY40)
    #define GDO0_PIN 14
#endif
#if defined(ARDUINO_TEENSY41)
    #define GDO0_PIN 24
#endif
```

Rodemfr, 2022-07-11:
Thanks ! I will merge it in master and close this ticket.
Meanwhile I think I discovered a new issue : MicronetToNMEA seems to crash after 1-2 hours running. I will open a new ticket.

[2 more replies]

closed 2022-07-11 by Rodemfr