Rodemfr/MicronetToNMEA #13: Reduce power consumption by setting CC1101 to low power mode

Rodemfr, 2022-07-14:
Currently, CC1101 is always listening to the radio even if network is only active less than 10% of the time.
A mecanism should be implemented to switch CC1101 to low power during idle phases of network.

Rodemfr, 2022-07-17:
I pushed the first version of "low power" version of MicronetToNMEA in "low_power_cc1101" branch. Its works fine here. I now use two features to reduce power consumption :
 - WFI instruction : this switch ARM CPU to low power mode until next interrupt
 - Disabling CC1101 XTAL outside of network window

The setup I used to test consumption is :
   - Teensy 3.5 board
   - CC1101 Board
   - M8N GPS Board
   - LSM303DLH Board
   - HC-06 Bluetooth board
   - DC-DC converter (12V->5V)

Results are :
   CPU@120MHz - No power saving  -> ~987mW / ~79ma@12.5V
   CPU@120MHz - WFI Only             -> ~812mW / ~65ma@12.5V
   CPU@120MHz - WFI & XTAL off    -> ~588mW / ~46ma@12.5V
   CPU@24MHz   - No power saving  -> ~750mW / ~60ma@12.5V
   CPU@24MHz   - WFI Only             -> ~712mW / ~57ma@12.5V
   **CPU@24MHz   - WFI & XTAL off    -> ~575mW / ~46ma@12.5V**

I can not measure power consumption on Teensy 4.0 but this could be a bit worse since WFI can not be used on it. It seems that time counters of Teensy 4.0 are handled in yield() function instead of interrupts like on Teensy 3.5.
WFI can disturb the system on Teensy 3.5 too, that's why I will continue testing before merging to master.

Rodemfr, 2022-07-17:
@dwarning & @j-lang it would be great if you could test this SW, especially if you can measure Power consumption.
If this version works well, I will number it 1.0 !

dwarning, 2022-07-18:
@Rodemfr : Which board should I use for testing? I see WFI is disabled for 4.x, is there a power saving for this boards too?

dwarning, 2022-07-18:
OK, but how to switch on/off the feature to make the compare?

Rodemfr, 2022-07-18:
I think I understood your readings : I commited WFI to master. So WFI was active in your master test. This explains the drop from 401.5 to 311.5. However, CC1101 does not change its current while it should. This could also be explained by the fact that WFI is disturbing timing measurements on Teensy 4, thus putting CC1101 in a full power mode.
Did you check that your display were working during your measurements ?

dwarning, 2022-07-18:
Not to misunderstand - only the USB current is measured the module currents are estimated. I can make with some effort separate measurements.
As I said, display was well in both setups.
I just seeing my pull of the master branch was not actual - it was from 15th July.
Sorry - I will repeat this run with your commits from 17th.

Rodemfr, 2022-07-18:
I finally merged everything in master to avoid enabling WFI on Teensy 4.x since it is causing troubles for me.
Now, I need to understand why you still have display with WFI on teensy 4 while I haven't.

Thanks anyway for taking the time to test.

dwarning, 2022-07-18:
Forgot to say - the only diff to master branch is my BoardConfig.h
[BoardConfig.h.txt](https://github.com/Rodemfr/MicronetToNMEA/files/9135604/BoardConfig.h.txt)
.

dwarning, 2022-07-19:
Two measurements with poor man's diff mode, a 2Ohm resistor in the usb power line. Both compiled with actual master code:
First the T3.6 with an average 115mA
[image]
Second the T4.1 with an average of 150mA
[image]

j-lang, 2022-07-21:
> @dwarning & @j-lang it would be great if you could test this SW, especially if you can measure Power consumption. If this version works well, I will number it 1.0 !

Tested current master and noticed it works great. 
Made a test by activating all GPS messages and the display still shows all values. 
Added a simple modification to M8Ndriver, added config for "at sea" according to: https://www.optimalsystem.de/os/docs/u-blox-gnss-konfigurationsanleitung.pdf

Rodemfr, 2022-07-23:
Thanks j-lang !
I merged your pull requests.
Note that I'm almost sure there is a sleeping bug in all SW versions which can make the software crash after some time. So keep an eye of potential display freeze with your setup.
I'm currently studying how to imprve my debug capabilities to solve it.

Rodemfr, 2022-07-23:
@dwarning : your measurements on USB power perfectly match the values given by my external power supply.