Rodemfr/MicronetToNMEA #19: Lowering SPI clock

dwarning, 2022-07-25:
I used lower SPI clock in freq_tracking branch 2MHz. With that no delays are needed in CC1101 operation.
Checked this with 3.6 and 4.1 board. In my opinion something is wrong in their 4.x SPI driver.
Make an side by side compare of the attached CC1101Driver file. 
True burst is not needed but gives few time saving in first FIFO access.
[CC1101Driver_cpp.txt](https://github.com/Rodemfr/MicronetToNMEA/files/9179289/CC1101Driver_cpp.txt)
[CC1101Driver_h.txt](https://github.com/Rodemfr/MicronetToNMEA/files/9179290/CC1101Driver_h.txt)

Rodemfr, 2022-10-16:
Using true burst is a good idea I think. I will make the change.
For lowering the SPI speed, it's more tricky because this increases the time of SPI transactions and thus reduces the CPU available since transfers are synchronous. I prefer to keep the SPI at 4MHz with dedicated delays.
There is something interesting in your modified CC1101 driver : you choose to use SpiStart/SpiEnd at each transfer instead of init/termination. Technically I think your way of doing is conforming to the spec, however, this as proven to make the porting on ESP32 fail because of the very long time of these function on the multicore platform. I don't really know what could be the best here. I still have in mind the idea to restart ESP32 porting now that we have a rewritten CC1101 Driver.

dwarning, 2022-10-16:
Oh, that's too old. Can't follow any more. Made few observations with scope in that time.

Regarding esp32 cc driver: Did you ever take a look into tty's code esp32micronet on github?

frgaude, 2022-10-16:
with regard to esp32. I took your work (master from a month ago) and tried to get the traffic scanning part to work...  and it works with very minor changes (I can't really code and I have no electronics background). 
my work is very rough (EEPROM not working, my settings are hardcoded, and I only kept the scanning part) but it is just to share that the radio/cc part *just* works on ESP32.

Rodemfr, 2022-10-17:
I think the most difficult part will probably be related to CC1101 transmission since it needs precise HW timer. esp32micronet projet should help with this matter.
The plan is to finish the work on the Teensy version (fixing bug, merging frequency shift algorithm) and then to go for the ESP32 port again.

closed 2022-10-17 by Rodemfr