Rodemfr/MicronetToNMEA #25: Problems with wind repeating feature

Rodemfr, 2022-10-28:
I observed while navigating that wind values were sometime staying stuck to static values. I suspect a problem with the wind repeating feature of MTN.
First, wind repetition should be made configurable in BoardConfig.h
Then, there should be two repeating modes : MTN acts as a second wind vane or MTN acts as a "master repeater" (i.e. like master do by default).
Finally, when acting as a second wind vane, wind values repeated by master device should be ignored to avoid possible overwrite of the real wind value by an older repeated value bouncing between master and MTN.

dwarning, 2022-10-28:
I saw no problems with wind values actualization. Can you describe your hardware setup to repeat the wrong behaviour.
For me: My Serial to Wifi converter is connected to MTN nmea serial port. The converter works as a AP. Wind vane is connected at port 6666 to this AP - also tablet / computer with navi software working on port 10110. Program (openCPN, avNav or qtVlm) listen on port 6666 and repeat it to port 10110. Any change from wind appears immediately on display.

Rodemfr, 2022-10-28:
My setup is the following : my wind vane is a tacktick one emitting directly to Micronet's network. It is mounted on top of the mast and all my displays are more or less in the shadow of the aluminium mast. Without MTN in the system, wind data is sometime not updated for several seconds because reception quality is on the low limit. In that case, display show dashes (--) on the AWS field.
To solve this I added a feature to MTN : it repeats wind data on the Micronet network. I fix MTN outside the mast shadow where wind reception is good and where connection to display is good also. This way, wind data is always received properly by my displays.
The problem I see is the following : my wind values (AXA/AWS) stay stuck but "without dashes", meaning that a value is periodically received but not changing. I suspect that depending on the order of response on the Micronet network, wind vane values can be ignored/overwritten by MTN. There is already a wind repeating feature in Micronet's master which I suspect to conflict with MTN's one. I need to do further tests.

Rodemfr, 2022-10-31:
I think I understood the problem. First here is the problematic system :
  - a wind vane on top of mast
  - several displays fixed to the cabin in the mast shadow with poor reception of wind values
  - MTN in the cabin, outside mast shadow, which receive has good reception of both wind vane and displays

Here is what happens : 
On the network, the talking order is : Display, then MTN and finally displays. Things go wrong when displays don't receive wind data from wind vane because of low RF quality. The faulty sequence is : 
  - New Micronet cycle
    - Master display repeats wind from the previous cycle -> OK, MTN captures this value
    - MTN repeats latest wind value (the one from master) -> OK, value captured by all displays
    - Wind vane sends new wind value -> Not seen by display because of poor RF, new value captured by MTN
  - End of micronet cycle : Display will show the value repeated by MTN, the one of the previous cycle

Things go wrong at the next cycle...

  - New Micronet cycle
    - Master display repeats wind from the previous cycle -> **NOK (this is value is two cycles old)**, MTN captures again this value, erasing the one received from wind vane at the previous cycle
    - MTN repeats latest wind value (the one from master) -> **NOK (this is value is two cycles old), value captured by all displays
    - Wind vane sends new wind value -> Not seen by display because of poor RF, new value captured by MTN
  - End of micronet cycle : Display will show the value repeated by MTN, the one of the previous cycle

The last wind value received by displays from wind vane will be repeated as long as wind vane will not be received again by displays. This can last 10-20s on my setup.

I see two solutions to this issue : 
1. When MTN is configured to repeat wind, it must not decode wind values repeated by master. This will solve the issue but will potentially delay the wind value by one cycle if wind vane is located behind MTN in the Micronet cycle.
2. Find a way to ensure that MTN will be last int he Micronet cycle, behind the wind vane. This might be possible by removing MTN slot and adding it again : to be verified...

dwarning, 2022-10-31:
Yes, I understood. Clear analysis. As I understand my constellation with wind over NMEA to MTN is not touched. That's the reason I have no problems. At the moment, where my wind wifi is in the near of the AP. May change next year if the wind vane is mounted on top.
You want / can not change the situation for better reception of the displays? Then you could filter out the MTN received wind values. Easy said - but not in practice.

closed 2022-10-31 by Rodemfr

dwarning, 2022-10-31:
I am a bit confused about terms and wording: Is it correct from my constellation with NMEA input from wind vane to speak about "Repeating"? I think I am using the original functionality of "Conversion".

May be other user will be confused too.

Rodemfr, 2022-11-01:
This feature is only for MTN to repeat wind data from Micronet's wind transducer to Micronet network again. Theoretically you don't need to do this, the wind transducer already sends the data to the network but the wireless tacktick wind transducer traditionnaly has RF problems because of its position on top of an aluminium mast. That's why I'm calling it a "repeating" feature. In your case, this feature is useless and can be safely disabled. MTN will simply convert the value coming from NMEA to Micronet.

Rodemfr, 2022-11-01:
But for what is confusion of the user regarding MTN configuration, I agree. I plan to add a dedicated chapter in the user manual with example configurations. It could be : 
- One full Micronet config with GNSS connected to MTN
- One Hybrid config (yours)
- Another Hybrid config (mine)
- etc.