Rodemfr/MicronetToNMEA #26: Use XDR instead of XDG

dwarning, 2022-10-28:
I searched for NMEA XDG sentence description and found nothing. But XDR is used with "V" specifier in openCPN manual for battery level.

Rodemfr, 2022-10-29:
Agree, I think I made a strange mix between HDG and XDR...
Looking deeper to OpenCPN doc I saw that Engine Dashboard plugin uses XDR with "BATTERY#0" name instead of "BATTERY" as MTN does today. So I will modify the sentence this way : 
`$INXDR,U,<voltage>,V,BATTERY#0,*<CS>`
This should make it visible to OpenCPN.

I will also replace HDG sentence by HDM. I noticed that dashboard plugin understands HDM and HDT but not HDG.

dwarning, 2022-10-29:
Regarding HDG - HDM: It is not well documented but qtVlm and avnav seems to decode both, openCPN only HDM.
It is preferable to use HDM to save some traffic because we haven't deviation and variation data available.
I will check it if you made the change.

dwarning, 2022-10-30:
I want add: All 3 investigated programs, opencpn, avnav and qtvlm show correct boat direction with the old HDG sentences. COG was 0 for this test case. opencpn's dashboard plug "GNSS Compass" show no reaction, this seems the mentioned problem. But as I said boat direction depends correct from HDG values.

BTW: I am using my old compass code. No success with your approach. Very large constant deviation of about 90deg.

Rodemfr, 2022-10-30:
Thanks for feedback. I'm back home now, I will fix it this week.
Concerning compass code, my compass is correct which means the problem you have is proper to LSM303DLHC (I have a DLH). I will order a DLHC to be sure to fix it properly.

dwarning, 2022-10-30:
> I will also replace HDG sentence by HDM. I noticed that dashboard plugin understands HDM and HDT but not HDG.

What is the problem with dashboard plugin? Can you explain.
If I activate "Mag HDG" in the plugin I see the values very well, same as the NMEA HDG values.
The "GNSS Compass" will only react with COG different from 0, I belief.

Rodemfr, 2022-10-31:
You are right for HDG vs HDM : no more problems with OpenCPN !?
I suppose I had something wrong with my setup when I tested... So I won't chagne HDG to HDM so far.

dwarning, 2022-10-31:
Yes, and just in case you need HDM or HDT for openCPN you can use the 
plugin NmeaConverter to generate the sentences.

Rodemfr, 2022-10-31:
Done

closed 2022-10-31 by Rodemfr