Rodemfr/MicronetToNMEA #28: Few code cleanings needed

dwarning, 2022-10-28:
CC1101Driver.cpp:50:15: warning: Member variable 'CC1101Driver::freqEstArray' is not initialized in the constructor. [uninitMemberVar]

MenuManager.cpp:82:14: style: Condition 'entry>0' is always true [knownConditionTrueFalse]
  if ((entry > 0) && (entry < menuLength))
             ^
MenuManager.cpp:79:9: note: Assuming that condition 'c>0x30' is not redundant
 if ((c > 0x30) && (c <= 0x39))
        ^
MenuManager.cpp:81:17: note: Assignment 'entry=c-0x30', assigned value is greater than 0
  int entry = c - 0x30;
                ^
MenuManager.cpp:82:14: note: Condition 'entry>0' is always true
  if ((entry > 0) && (entry < menuLength))
             ^

MicronetMessageFifo.cpp:53:22: warning: Member variable 'MicronetMessageFifo::store' is not initialized in the constructor. [uninitMemberVar]

MicronetSlaveDevice.cpp:56:22: warning: Member variable 'MicronetSlaveDevice::networkMap' is not initialized in the constructor. [uninitMemberVar]

 
Quote [Rep](https://forums.ybw.com/index.php?conversations/micronettonmea.598931/reply&quote=650702)

Rodemfr, 2022-11-04:
I installed cppcheck on windows and it gives me differents warnings than you, some are really unrelevant for this type of SW. Yours are better. Are you using it on Linux ? Which version ?

dwarning, 2022-11-04:
I use it on Ubuntu. Version is 2.7

My command:

  cppcheck --enable=all *.cpp

Rodemfr, 2022-11-05:
Done !

dwarning, 2022-11-05:
Hi Ronan,
great work to clean the things.

Last one remaining from the complicated cast in RFDriver.cpp, gotten fro Arduino-IDE:

/home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetToNMEA/RfDriver.cpp: In member function 'void RfDriver::TransmitCallback()':
/home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetToNMEA/RfDriver.cpp:408:97: warning: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
   cc1101Driver.WriteTxFifo(static_cast<const uint8_t*>(preambleAndSync), sizeof(preambleAndSync));
                                                                                                 ^
In file included from /home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetToNMEA/RfDriver.h:13:0,
                 from /home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetToNMEA/RfDriver.cpp:8:
/home/dietmar/Projects/Arduino/MicronetToNMEA/MicronetToNMEA/CC1101Driver.h:150:7: note:   initializing argument 1 of 'void CC1101Driver::WriteTxFifo(uint8_t*, int)'
  void WriteTxFifo(uint8_t *buffer, int nbBytes);

dwarning, 2022-11-05:
I resigned to ++ overloading functionality for write fifo function as in my branch true_burst:
`	void WriteTxFifo(uint8_t data);
	void WriteArrayTxFifo(uint8_t *buffer, int nbBytes);
`
No warnings anymore and much easier to understand for ordinary programmer like me.

Rodemfr, 2022-11-05:
Fixed. I will wait for you confirmation this time before closing...

dwarning, 2022-11-05:
Now its better, but why in heaven you using this terrible cast in RfDriver.cpp:

`		cc1101Driver.WriteArrayTxFifo(static_cast<const uint8_t*>(preambleAndSync), sizeof(preambleAndSync));
`
using (uint8_t *) is sufficient for me. Is it correct?

Rodemfr, 2022-11-05:
Because my ccpcheck version asks for this. Be sure I would not have done this by myself :-)

closed 2022-11-05 by Rodemfr

dwarning, 2022-11-05:
But now you have the separate function for array transfer and no 
overloading anymore. The complicated cast is not needed.