Rodemfr/MicronetToNMEA #29: Incorrectheading values with LSM303DLHC

Rodemfr, 2022-10-30:
dwarning sees heading values with very large offsets (around 90°). He is using a LSM303DLHC which uses a different driver than LSM303DLH.

Rodemfr, 2022-11-12:
My order of LSM303DLHC has just been cancelled :-(
I have to reorder it...

frgaude, 2022-11-13:
I have one I can send you.

Rodemfr, 2022-11-13:
Thanks ! No need, I ordered a new one from another source. I should get it newt week.

Rodemfr, 2022-11-15:
I received the board. I will not be able to work on it before next week however.

Rodemfr, 2022-12-01:
I soldered a LSM303DLHC and my heading is correct. I don't see the offset you have.
When X axis is pointing to the north I read zero.

tvr256, 2022-12-01:
I also have a LSM303DLHC, and something is not right somewhere:

If I point the compass X axis (very) roughly north, I get the following NMEA output from MTN: `$INHDG,582,,,,*5f`. Meanwhile, my TackTick dual display shows 300 degrees.

The Tacktick display has calibration set to `HDG -167 VAR W02 DEV ---` (from my previous compass).  
And in MTN, the calibration settings are : `Magnetometer calibration : -98.00 -73.00 268.50`

tvr256, 2022-12-02:
One other thing I noticed, menu 1 shows the wind and water offsets (presumably read from Micronet) but it doesn't show the compass offsets. How easy would it be to add them?

```
MicronetToNMEA, Version 1.3
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -98.00 -73.00 268.50
```

dwarning, 2022-12-02:
Do you made the procedure with rotating and swinging round the board to come in all directions - point 7 in menu? I think so - you have shown values. After that you should errors < 5°. Have in mind that the sensor is very sensitive for metal and electronic equipment. I made the compensation in the garden.

In my opinion the setting of compass offset in the Micronet display is for a kind of fine tuning to adapt the compass to your special situation in the boat. They should not interfere with the basic sensor calibration.

tvr256, 2022-12-02:
> Do you made the procedure with rotating and swinging round the board to come in all directions - point 7 in menu? I think so - you have shown values. After that you should errors < 5°. Have in mind [...]

Yes, I followed the instructions in the user manual [here](https://github.com/Rodemfr/MicronetToNMEA/blob/master/doc/user_manual/user_manual.md#calibrating-navigation-compass). The calibration values seem quite extreme, but the amplitude values are very close. The calibration was done at my desk, so I'll try again outside.

> In my opinion the setting of compass offset in the Micronet display is for a kind of fine tuning to adapt the compass to your special situation in the boat. They should not interfere with the basic sensor calibration.

The offset is important when the compass is mounted without the X-axis pointing directly ahead. The NMEA HDG sentence and the Micronet display should both always include the compass offset.

In my test, the NMEA HDG sentence says 582 degrees which really doesn't seem right - I'd expect it to be between 0 and 360 degrees. Meanwhile the displayed bearing is 300 degrees, which also seems wrong.

Rodemfr, 2022-12-02:
For the `$INHDG,582,,,,*5f` error, this looks like a modulo error. Your large heading offset of -167 emphasizes an error when I add the offset to the heading value : I forgot to check for 0/360° limits, leading to stupid values like 582. It will be fixed tonight.
Concerning heading offset, it is correctly stored in EEPROM by MTN but not printed in menu 1. I will add it to menu, it's very easy.

Concerning LSM303DLHC, that's the first time I use it and I discover that the scale of values is not the same than for my former LSM303DLH. Values are now around 1000 while previously around 1. I will read the datasheet but there must be something like a 10 bit scale somewhere (1024 factor). Technically, it does not change the angle calculation but it will better match the examples given in user manual. Be careful that once I will push the modifications into the code, you will have to recalibrate magnetometer.

A last point, I observed that the amplitude of X,Y ans Z axis are not exactly the same when measuring the same magnetic field. X & Y are identical, while Z is slightly smaller. I remember of something mentioning this in datasheet. It would improve precision of angle calculation to compensate for this small variation of amplitude on Z axis. I will also add this to the code.

dwarning, 2022-12-02:
@Rodemfr : In case you make a rework for DLHC driver.
LSB_per_Gauss_ and mGal_per_LSB can removed from both files.
I think we want not know the temperature in the sensor. So you can write 0x10 instead of 0x90 into CRA_REG_M.
Other settings are same as mine.

Rodemfr, 2022-12-02:
@tvr256 : I pulled a fixed in branch `fix_issue_#23`. This should solve your heading problems. I scaled down the values of magnetic fields to gauss units instead of raw readings. I think found the issue of `INHDG` sentence. Heading offset was applied twice... Let me know if this is OK on your side.

@dwarning : I disabled temperature heading and removed mGal_per_LSB. However I kept LSB_per_Gauss_XYZ because I use it to scale down raw values of magnetic field and also because the different sensivity between XY and Z axis are corrected by this operation.

tvr256, 2022-12-02:
Well it's different, but still strange... I've kept my offset at -167 degrees.

Pointing the X-axis (roughly) north : Micronet displays 300 degrees, NMEA HDG shows 40 degrees
Pointing the X-axis (roughly) west : Micronet displays 226 degrees, NMEA HDG shows 35 degrees
Pointing the X-axis (roughly) south : Micronet displays 228 degrees, NMEA HDG shows 37 degrees
Pointing the X-axis (roughly) east : Micronet displays 231 degrees, NMEA HDG shows 39 degrees

I think maybe the X, Y and Z axes have got  up muddled up somewhere?

tvr256, 2022-12-02:
Just realised I forgot to recalibrate! I'll try again now...

Rodemfr, 2022-12-02:
Making more tests I already found another issue... MTN can send invalid values to Micronet, leading to the display of ` ---`

tvr256, 2022-12-02:
Okay, so after calibrating it's far better but still not quite right. With offset -167:

With x axis pointing roughly north, Micronet displays 359 degrees, NMEA HDG shows 168 degrees. It seems like Micronet might be ignoring the heading offset, while NMEA is adding instead of subtracting?

Also, the offset isn't displayed correctly in the settings:
```
MicronetToNMEA, Version 1.3
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -0.08 -0.06 0.27
Heading offset = 4
Magnetic variation = -3
```
The displayed magnetic variation is correct (3 degrees W), but for some reason Heading offset is displayed as 4, not -167.

Rodemfr, 2022-12-02:
Strange ! I continue investigation ...

Rodemfr, 2022-12-02:
I just pushed a new fix. I used the same values than you : -167 & -3. This works for me.
I don't really explain the value of `4` you have instead of `-167' could this be that you configured your heading offset while MTN was not in *NMEA conversion mode* ? MTN only catch configuration changes when in this mode.

tvr256, 2022-12-02:
Yes, the -167 offset is left over from my previous compass installation, I haven't changed it since using MTN.

I'll be busy for the rest of this evening, but I'll test as soon as I'm free. Thanks for the quick fixes!

Rodemfr, 2022-12-02:
No problem. Let me know the result for me to close the issue.

tvr256, 2022-12-02:
After re-saving my offset at -167, it's now displayed correctly:

```
MicronetToNMEA, Version 1.3
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -0.08 -0.06 0.27
Heading offset = -167
Magnetic variation = -3
```

And now with the x axis roughy north, Micronet displays 4 degrees and NMEA HDG says `$INHDG,2,,,,*52`. Would you expect them to show the same value?

Rodemfr, 2022-12-02:
Yes, I would expect the value to different of 1 max (float rounding). Diving into the code ...

tvr256, 2022-12-02:
It also seems strange that my compass is showing north when the x-axis points roughly north. Id expect it to be out by 167 degrees one way or another. Or is my compass offset by the same amount as a complete fluke?

Rodemfr, 2022-12-02:
I found a serious error in my handling of heading offset. What is strange is that your error between HDG sentence & your Micronet display is so small... I pushed a new fix. This time you don't need to calibrate again.

Rodemfr, 2022-12-02:
If I understand well the error I made, you will probably have to re-estimate your heading offset of -167 on your boat. You will probably find +167 this time...

dwarning, 2022-12-03:
I made a check on branch issue_#29 with DLHC.
It works so far but accuracy is worse. Maybe it has to do with the reduced digit count for the coefficients in case it is not only the printf format.
I this context I have a big request: Can we agree to use LSB counts as offset coefficients and make scaling and number conditioning later? You know I make experiments from time to time with different algorithms. And I mean the offsets are only dependent from sensor IC's and board assembly situation. With LSB's as coefficients we have a universal interface for DLH/DLHC and any user can build his own algorithm if he want. I see no disadvantage to do that. Correct me if I am wrong.

Rodemfr, 2022-12-03:
What do you mean by "accuracy is worse" the scaling down being done in floating point there should not be any accuracy loss here.
Concerning the scaling down itlself, we need it because the sensitivity of Z axis is not the same than for X and Y. So by dividing with these sensitivity constants given by datasheet, we balance the 3 axis so that the heading calculation algorithm will work properly without taking care it self of axis differences. Said in another way, with scaling, magnetic field will move on a sphere while unscaled value will move on a ellipsoid, giving slightly wrong results when boat will roll.
I think this difference of sensitivity on Z axis is due to restrictions in IC building o the Z axis, I imagine that you cannot have layers as thick as you want, or something like that.
What I can do is to add a new magnetic reading function called `GetRawMagneticField` which would return the raw integer readings from the chip so that you could use it in your own algorithm.

tvr256, 2022-12-03:
Finally had a chance to test again. Offset set to -167 in Micronet, and confirmed in MTN config.

X axis pointing roughly north : Micronet displays 186 degrees, NMEA HDG says `$INHDG,354,,,,*52`.

354-167=187. So it looks like the offset is being correctly applied by Micronet, but not in the NMEA sentence?

tvr256, 2022-12-03:
So this is really screwy. In my previous post, after flashing the latest code my offset in MTN had reverted from -167 to 4 for some reason. Is it being saved correctly?

So I've changed offset back to -167 in Micronet and confirmed it's updated correctly in MTN config. Now with X axis north, Micronet is displaying "---" and NMEA says `$INHDG,355,,,,*53`. Looks like MTN still isn't applying offset correctly, and Micronet is now receiving an invalid value?

dwarning, 2022-12-03:
Regarding accuracy to be more specific: I checked the four directions and have about 5 ... 14° deviation. I would withdraw my wording _worse_. Perhaps it is more my specific worse setup.
Anything you said about taking care for different axis sensitivities is right but you can do it later in your specific algorithm too. In the beginning we are reading LSB's and they must corrected by offset. Anything else should placed in the algorithm.
The raw readings would be a compromise but  should be placed in a convenient namespace so an exchange of NavCompass.* is sufficient.

Rodemfr, 2022-12-03:
> So this is really screwy. In my previous post, after flashing the latest code my offset in MTN had reverted from -167 to 4 for some reason. Is it being saved correctly?
> 
> So I changed offset to -166 in Micronet and then back to -167. And it's now showing as 749 in my MTN config:
> 
> ```
> MicronetToNMEA, Version 1.3
> Device ID : 3123456
> Attached to Micronet Network 810801CB
> RF Frequency offset = 8.51 kHz (9 ppm)
> Wind speed factor = 1.00
> Wind direction offset = 0
> Water speed factor = 1.00
> Water temperature offset = 0
> Using LSM303DLHC for magnetic heading
> Magnetometer calibration : -0.08 -0.06 0.27
> Heading offset = 749
> Magnetic variation = -3
> ```
> 
> And now with X axis north, Micronet displays heading as "---", while NMEA says `$INHDG,355,,,,*53`

Hum, really strange ! There must be something I'm missing somewhere. Diving again in the code and trying to get the same problem than you...

Rodemfr, 2022-12-03:
It is useful indeed. When I set -167 on my Micronet display, I get -167 in the info menu, where you have 749. This is where the different behavior is between your MTN and mine. To solve this, I need you to answer a few questions :
- What display are you using to set the heading offset ? 
- What is your Teensy model ?

Also, there is something you can do to compare your system to mine : put MTN into Traffic scanning mode (menu 5) and then use your display to set heading offset to -167. You will see the configuration message pass on the console. My MTN shows this message :
`83092C36 81037082 06 09 00 73 13 13  -- FF 09 02 59 FF 09 6B  (-38, 39022)`
This is the N°6 message used by the display to tell the new heading offset. Could you capture yours and post it here ?

tvr256, 2022-12-03:
I'm using a TackTick T111 dual display, and a Teensy 4.0.

The first time I updated the offset it worked fine, so it's obviously an intermittent problem.

I'm out at the moment, but I'll run some tests as soon as I'm home and try to recreate it with some debug messages.

Rodemfr, 2022-12-03:
no problem, there is no urgency. You have the same config than me. So you will likely see the same values in the configuration message. It is however good to confirm it. I now seeking what could provoke such a change in this offset value.

Rodemfr, 2022-12-03:
I pushed a small modification which prints the current configuration at startup, before starting conversion. It will reveal if the value changes later during SW execution.

tvr256, 2022-12-03:
Start : MTN offset 749, Micronet offset -167

```
MicronetToNMEA, Version 1.3
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -0.08 -0.06 0.27
Heading offset = 749
Magnetic variation = -3
```

Change offset in Micronet to -166:

`810801CB 810801CB 06 01 00 B1 13 13  -- FF 09 02 5A FF 05 68  (-51, 20324)`

Immediately change back to -167:

`810801CB 810801CB 06 01 00 B1 13 13  -- FF 09 02 59 FF 06 68  (-49, 20324)`

Check MTN offset, still set to 749!

```
MicronetToNMEA, Version 1.3
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -0.08 -0.06 0.27
Heading offset = 749
Magnetic variation = -3
```

tvr256, 2022-12-03:
Testing again, the MTN offset didn't change because I was in  traffic scanning mode (5). I switched back to NMEA conversion mode (4) and a changed offset is now saved correctly in MTN. I can't get it to misbehave again, so I have no idea how it changed to 749.

tvr256, 2022-12-03:
After some more testing, offset seems to be working perfectly now.

But there is one final small issue... In Main.cpp line 672, the magnetic variation is added onto the magnetic heading. This is incorrect, magnetic variation should be completely ignored by MTN. It should only be used internally by Micronet, to calculate true heading.

Currently in the UK, magnetic variation is only +0.4 degrees so it makes practically no difference in the real world. But it would be nice to make it perfect :)

tvr256, 2022-12-04:
Thinking about it some more, an even better solution would be for MTN to read both variation and deviation from Micronet and add them to the HDG sentence.

However, the TackTick documentation states that the variation setting "is ignored if variation information is available from a GPS source". I don't think the M8N sends variation in its RMC sentence (mine's not wired up so I can't check), but the ultimate solution would be for MTN to calculate variation based on M8N lat/long, using [a similar algorithm to SignalK](https://github.com/SignalK/signalk-derived-data/blob/master/calcs/magneticVariation.js). Or would that be complete overkill?

Rodemfr, 2022-12-04:
Thanks a lot for testing all of this, that's useful ! My boat now has a magnetic compass in the tiller pilot, connected to my micronet network, so that I'm never testing this part of MTN.
Concerning line 672 of Main.cpp : you are correct. I fixed it in the branch.
Concerning your proposal to go deeper in compass handling, it is worth to look at, I will enter a new ticket in Github to start the discussion.
If you could just test one last time the branch, I will merge it into the master.

dwarning, 2022-12-04:
I have still problems with the new code and my DLHC/T4.1. companion.
I made different compensations, some with z-axis only to top (more realistic for a boat) and some with z-axis to top and bottom. How you handle it?
This comes from only to top compensation (I used longer printf format):
Magnetometer calibration : -0.0459  -0.1177  0.0010
Heading offset = 0
Magnetic variation = 0

Can you both show me your coefficients?
In different trials best max deviation was around +/- 10° in the four directions.
But: Tilt compensation did never worked. My setup is on a development board. I make tilts of about 33° and get more then 50..70 deg deviation which stands stable, Sometime backboard is OK, and starboard total wrong with 80° dev. dependent from direction.

Will check it with my DLHC/T3.6 board next.

How are your experiences?

tvr256, 2022-12-04:
I hadn't even thought of testing tilt compensation. Why is heading so complicated?

I've tested quickly this morning, and offset variation and deviation seem to be working as expected.

I'll review the code to make sure there are no more hidden errors. To make the code more readable (and prevent accidental errors in the future) it would be helpful to explicitly mention which heading is being used everywhere in the code. For example, GetHeading could be renamed to GetCompassHeading.

And just because I haven't seen it explicitly written anywhere in this conversation or the code, here's how we should be calculating values:

1) Compass Heading = X + tilt compensation (Y, Z)
2) Magnetic Heading = Compass Heading + Offset + Deviation
3) True Heading = Magnetic Heading + Variation

tvr256, 2022-12-04:
If we're going to go down the rabbit hole of tilt compensation algorithms, Pololu has already done all the hard work so we don't have to!

https://github.com/pololu/lsm303-arduino

tvr256, 2022-12-04:
Unfortunately it's happened again - I flashed the latest build, then started MTN, then started up Micronet, and my heading offset has changed from 0 to 8452:

```
MicronetToNMEA, Version 1.3
Valid configuration found in EEPROM
Device ID : 3123456
Attached to Micronet Network 810801CB
RF Frequency offset = 8.51 kHz (9 ppm)
Wind speed factor = 1.00
Wind direction offset = 0
Water speed factor = 1.00
Water temperature offset = 0
Using LSM303DLHC for magnetic heading
Magnetometer calibration : -0.07 -0.06 0.27
Heading offset = 8452
Magnetic variation = 0
```

Unfortunately I didn't capture the config before starting MTN, so I don't know whether it was the flashing process or a rogue message from Micronet. I'll keep testing until I can narrow it down.

Rodemfr, 2022-12-04:
> I have still problems with the new code and my DLHC/T4.1. companion. I made different compensations, some with z-axis only to top (more realistic for a boat) and some with z-axis to top and bottom. [...]

Here are my offsets : 0.01 -0.10 0.07
Heading offset and variation are at 0.
I tested very roughly tilt compensation by hand, it is roughly working but with what you are reporting I need to make more serious tests. I wrote myself the formula for fun so it might be flawed... I will engage more precise tests with a fine calibration and a tool to rotate precisely, far from any metal or electric object.
I will create a ticket dedicated to the tilt issue.

Rodemfr, 2022-12-04:
> Unfortunately it's happened again - I flashed the latest build, then started MTN, then started up Micronet, and my heading offset has changed from 0 to 8452:
> 
> ```
> MicronetToNMEA, Version 1.3
> Valid configuration found in EEPROM
> Device ID : 3123456
> Attached to Micronet Network 810801CB
> RF Frequency offset = 8.51 kHz (9 ppm)
> Wind speed factor = 1.00
> Wind direction offset = 0
> Water speed factor = 1.00
> Water temperature offset = 0
> Using LSM303DLHC for magnetic heading
> Magnetometer calibration : -0.07 -0.06 0.27
> Heading offset = 8452
> Magnetic variation = 0
> ```
> 
> Unfortunately I didn't capture the config before starting Micronet, so I don't know whether it was the flashing process or a rogue message from Micronet. I'll keep testing until I can narrow it down.

This is once again very strange. All your configuration parameters have been recovered from EEPROM, except heading offset. It means the checksum of the EEPROM block was valid. So it means that at sometime, this value of 8452 has been properly written in EEPROM by Configuration class with the correct checksum. I conclude that the heading offset has probably been overwritten during runtime before beeing saved again into EEPROM. In NMEA conversion mode, MTN automatically save new configuration data to EEPROM if any of the parameter changes.
If you switched between various versions of SW after having set heading offset to -167, could this be that the older master version without the heading offset fix could generate such a silly value ? Maybe, but I fear something much more difficult to find. Also, I'm still wondering why you have this issue and not me...
I merged the branch into master because it is obviously less buggy than before but I let the ticket open until we find your problem.

tvr256, 2022-12-04:
I have a sneaking suspicion it might be related to the Deviation menu. I've only seen the error twice, and both times it happened some time after I had tried (and failed) to set the deviation. I'll keep testing with logs enabled, I'm sure I'll find the root cause eventually.

tvr256, 2022-12-05:
I've just found another small issue with compass values. In DataBridge.cpp line 842, magnetic variation is written to the wrong field in the NMEA HDG sentence.

$INHDG,23,0,E,,*14 should read $INHDG,23,,,0,E*14.

Rodemfr, 2022-12-06:
I found an error in the code which could potentially explain your heading offset issue (and others).
Heading offset is transmitted with command 0x06 (Set Parameter) on page 0xff and Parameter ID 0x09. The code of MicronetCodec.cpp checks the Parameter ID but not the page !! It means that if one of your device writes Parameter ID 0x09 on any other page, heading offset will be overwritten.
I won't be able to fix this today but hopefully tomorrow. I also also fix your latest finding on HDG.

Rodemfr, 2022-12-07:
I fixed the parameter handling bug as well as the HDG syntax issue. I made an error in my git handling so that the fix has been pushed directly into master...
You can test it. I think it has good chances to solve the problem. I have seen in my MTN a water speed factor of 2.80 (!) which I never entered myself. This really looks like what you have with Heading offset and can be pretty well explained by the error I found in the code.
Let me know the result.

tvr256, 2022-12-07:
And suddenly it all makes sense! In issue #62 I reported that I'm receiving an undocumented `F9 09` message from the network. MTN was incorrectly interpreting this as heading offset ( `FF 09`) and overwriting the value.

I'm away from home so probably can't test until the weekend, but I'm pretty confident you've found the root cause. Great detective work!

closed 2022-12-12 by Rodemfr