Rodemfr/MicronetToNMEA #37: actual git not working for T3.6

dwarning, 2022-11-04:
Made checks with T3.6 and T4.1. Last one is fine. But T3.6 shows nothing on display.
Will check the introduced burst things at first.
Frequency is OK, Signal strength strong.
Can I deactivate the new frequency approach?

Rodemfr, 2022-11-04:
Yes, you can disable it by commenting out line 238 : cc1101Driver.UpdateFreqOffset(); of file RfDriver.cpp

Rodemfr, 2022-11-04:
I just tested with my Teensy 3.5 : I have the same issue. It seems commit 10572d0 "Remove polling from RX ISR" created the problem.

Rodemfr, 2022-11-04:
I founf the issue : Teensy seems not to like interrupts configured to be triggered on "HIGH" level. I had to modify the attachInterrupt line in MicronetToNMEA.ino by this : 
```
#if defined(ARDUINO_TEENSY35) || defined(ARDUINO_TEENSY36)
	attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, RISING);
#else
	attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, HIGH);
#endif
```

Strange, I don't really explain why...

dwarning, 2022-11-04:
Ronan,
the failure is gone with your last commit.
Fabulous work.

dwarning, 2022-11-04:
Works now with T3.6.

closed 2022-11-04 by dwarning