Rodemfr/MicronetToNMEA #43: Timing evaluation for spi burst transfer

dwarning, 2022-11-06:
I settled my setup for scope and rf transmission (rtlsdr) measurements. I a first view I see very compact transmission chunks with large gaps between.
Do you have an idea which time - signal constellation are of special interest?

dwarning, 2022-11-06:
This are a first trial. The problem is I trigger to CH4, If it is CS then decoding is OK, if it is gdo0 the decoding is wrong. But to the the delay it would be OK.

Sure, I can use mosi or miso in different situation but the only one decoding is OK.

An overview with CS on channel 4:

[image]

An overview with GDO0 on channel 4:

[image]

Last one zoomed to the first GDO0 rise.

[image]

Rodemfr, 2022-11-06:
This is interesting. We see the first two SPI bytes corresponding to the request for the FIFO level : 1 address byte, one byte for the level. Then the second long request corresponds to the reading of FIFO data. We have 17 bytes exhanged : 1 address byte, and 16 bytes corresponding to the RX FIFO level of 16. The next two byte transfer 3-4us later is probably the programming of the packet length.
What speed is running your CPU, 600MHz or 24MHz ?

Rodemfr, 2022-11-06:
I also see that the 8us delay we have between CS High to Low transistions is damaging the overall timing. We need this delay to avoid overflowing CC1101 but I doubt this is required for every command. It could be worth to investigate command per command if we can remove this delay or at least reduce it to 1us instead of 8.

dwarning, 2022-11-06:
Don't know where you see 8us between CS High to Low transition.

But you understood that I exchanged source for ch4 between pictures? Switched from CS to GNDO0.

Don't know about CPU speed. This was made with T4.1 and unchanged code from freq_tracking branch. SPI clock I measured 4MHz.

Rodemfr, 2022-11-06:
The 8us wait between CS High and Low is in the code. I don't se CS in the last picture but I know where it is pulled down and released high. I don't know how much time there is between the first and second SPI transfers but it includes this 8us wait.

dwarning, 2022-11-06:
Few CS periods I have gathered. Last one shows only 7us instead of 8. Is that what you meant by "command by command "?
[image]
[image]
[image]
[image]
[image]

Rodemfr, 2022-11-06:
Yes. Each time a "command" is sent to CC1101Driver (e.g. read FIFO level, or read FIFO bytes, etc.) CC1101 driver pulls CS low and releases it after the end of the SPI transaction. Then, at the next command (i.e. the next CS low/high cycle) it ensures that 8us have been elapsed since the previous command and waits the necessary time if not.
This has been done this way because if we don't wait these 8us, CC1101 does not go fast enough to handle the next command on time and RF communication gets broken. The problem is that we wait 8us for every command. I'm pretty sure that simple commands like GetFifoLevel() do not require any wait at all. So to continue to optimize CPU load and ISR execution time, I should test command type per command type what delay is necessary behind CS high to let time to CC1101's FW to process it.
Datasheet mentions delays after CS in nanoseconds but obviously one of the command needs more than that.

Rodemfr, 2022-11-06:
By the way, your measurement of 7us instead of 8 pointed me to a bug in the code : a '<' comparison instead of a '<=' !

dwarning, 2022-11-09:
Could you be so kind to give away the secret, meant where is the bug.
Or should I mark it as an issue?

Rodemfr, 2022-11-10:
The 8us wait is operated in ChipSelect() and ChipDeselect() function of CC1101Driver.cpp.
ChipDeselect() stores the times in lastCSHigh variable just after having set CS high. Then at the next ChipSelect(), the code ensures that DELAY_BETWEEN_CS_US has elapsed since the previous Deselct and waits in a loop if necessary.
I added this when j-lang was facing troubles running for the first time MicronetToNMEA on Teensy 4.0@600MHz. I discovered that Teensy was too fast for CC1101 abnd that we had to let time for it to process the commands. However I coded a wait for every SPI transfer while it is very likely that only one or two specific commands need this wait.

Rodemfr, 2022-11-10:
It is not a bug, but it is an optimization to do : investigate which command needs which delay and only wait 7-8us for these commands.

dwarning, 2022-11-10:
Made an empirical check: ChipSelect with a parameter for delay instead DELAY_BETWEEN_CS_US. Only about 7 functions in CC1101Driver.cpp are using this - not GetFifoLevel() etc. or other.
Outcome was 6us was anytime sufficient for stable display, few functions can live with 4us. But can not found the bottleneck function.

Interesting: Instead of the us - loop elechouse make pinMode() and SPI.begin before any transfer. This seems to work independent from Teensy's speed condition. But in my opinion it is a bit unlogical.

Rodemfr, 2022-11-10:
Agree for pinMode and SPI.begin. Also SPI.begin execution time can vary a lot depending on platform (I remember during the ESP32 port that this function was very long).
What we can do is to let about 7us for all functions except for FIFO related ones. It will speed up a lot RX & TX ISR.

closed 2022-11-12 by Rodemfr