Rodemfr/MicronetToNMEA #44: Cleaning up

dwarning, 2022-11-08:
Not an issue.
Because you are just on the way:
Can we shift the LSM303DLHx register macros from *.cpp to *,h? Or are reasons to do it here?
What is happen with NavCompassDriver.cpp, it is empty.

Rodemfr, 2022-11-08:
That's a convention I usually follow : don't expose internal definitions in header files that are supposed to be an external interface (external to LSM303 driver). In normal times I would create a dedicated header file called "LSM303DLH_Registers.h" or something like that, but since I need to keep all files in the same folder to keep compilation with Arduino IDE easy, I avoid creating too many files when I can.

Rodemfr, 2022-11-08:
By the way, I opened the possibility to have "discussions" on the project homepage for every type of discussions other than issue. We can use it for this. I will move the issue there.