Rodemfr/MicronetToNMEA #47: Use FSCTRL0/FREQOFF to calibrate XTAL

Rodemfr, 2022-11-12:
Currently, MicronetToNMEA calibrates PLL frequency by directly shifting PLL's frequency control word (_FREQ0,1,2_). This works well but it would be better to calibrate by letting _FREQ0,1,2_ to the theoretical frequency and then to use _FREQOFF_ to find master's frequency. This was the initial purpose of _FREQOFF_ register.

dwarning, 2022-11-12:
I am absolute agree. 
Wouldn't speak about _FREQOFF_ register, use _FSCTRL0_ instead with FREQOFF value.

Rodemfr, 2022-12-04:
There are still many tickets to solve on MTN so I will save my time by not solving this one. The current frequency calibration works well, its only that I'm not using the CC1101 in the most beautiful way.

closed 2022-12-04 by Rodemfr