Rodemfr/MicronetToNMEA #50: HDG info not sent to Micronet?

frgaude, 2022-11-14:
Hi Rodem, there seems to be a regression with the lastest batch of commits:
I used to get the heading info on my displays, it's no longer showing with current "master". 
the HDG info still shows in the NMEA feed.

same behavior with or without compass calibration.

GPS COG & SOG are displayed ok.

[NMEA.log](https://github.com/Rodemfr/MicronetToNMEA/files/10005579/NMEA.log)
[image]

Rodemfr, 2022-11-14:
I just made changes on this area in the latest commits, I will check it tonight. It should be easy to fix.

Rodemfr, 2022-11-14:
Mag heading is working on my setup. When I look to the modifications I have done, I suspect a possible problem.
First, I found a few problems but I made an error with my git and I pushed the fix directly in master... So you can test it directly in master branch.
Secondly, If it does not work, you can also reset the heading offset and magnetic variation with your display or remote (in compass option page). Even if these two values are already set at zero, it is important to reset them again so that MicronetToNMEA can see them pass on the network and store it into EEPROM.
Let me know the result.

dwarning, 2022-11-14:
Only 1 cent from my side:
Heading self is working with latest master, see the correct nmea sentences. But no display values.
All calibration settings disappears and setted to 0. Also for different older version. Perhaps the EEPROM content/addresses changed.

frgaude, 2022-11-14:
still no luck with the latest changes.

I'll revert to an older version to double check.
I might be able to publish some micronet logs from a 2nd board (an esp32 with a cc1101 :-) )

And to clarify my situation with magnetic calibration: until this evening, it was set to 0 on the various elements I have at home (MTN board and the two displays are not yet in use on the boat). 

I did a calibration after I experienced the issue. to be in a more "by-the-book" situation. 
 
I'll reset the calibration.

thanks for the great support!

frgaude, 2022-11-14:
I went back to the merge of pull request #42 and ... the heading info is displayed.

one thing to note: in this version and versions prior to the rework on the RF driver, the calibration of my CC1101 board was really skewed to the low end of the spectrum:

`
.*******..............................................................................................
Frequency = 869701.94kHz
Range = 17.94kHz
Deviation to real frequency = -138.06kHz
Do you want to save the new RF calibration values (y/n) ?
Configuration saved
`

RF calibration is more "centered" with the latest code:

`.......................************************.......................................................
Frequency = 869793.19kHz
Range = 68.79kHz
Deviation to real frequency = -46.81kHz (-53 ppm)
Do you want to save the new RF calibration values (y/n) ?
Configuration saved
`

so maybe, somehow, my CC1101 is causing this...

frgaude, 2022-11-14:
pinpointed to this commit (works before, stops working after):
[image]

Using a teensy 3.5 + CC1101 + LSM303DLHC + GNSS M8N (+ Wifi bridge deactivated for these tests)

frgaude, 2022-11-15:
maybe the # of paired devices:

[image]

vs

[image]

dwarning, 2022-11-15:
Code recover is not easy because @Rodemfr mixed code changes with commenting and formatting in same commits.

Similar setup:
teensy 4.1 + CC1101 + LSM303DLHC + GNSS M8N (+ Wifi bridge not used) + 1x MN 100-2

Playback to commit 97819d was not sufficient to get headings on display. 
Can confirm that playback to commit 598f02 brings the hdg values back. Same as your results.

frgaude, 2022-11-15:
found it :-) !

[image]

I'm sincerly impressed by the work of Rodemfr, clean code, refactoring whenever he feels it's needed, deep diving into the intricacies of the radio datasheets to get everyting right, great documentation, etc... 

So spotting the ittle copy paste errors here and there is the least I can do (and also the only thing I'm able ti do :-) )
 
I'm totally fine to have to dig into

Rodemfr, 2022-11-15:
Excellent work ! I will look at it calmly tonight and review a bit what I did...

Rodemfr, 2022-11-15:
If you confirm me that it is now OK with this issue, I will close it.

frgaude, 2022-11-15:
Yes working fine now!

closed 2022-11-15 by Rodemfr

Rodemfr, 2022-11-16:
> I went back to the merge of pull request #42 and ... the magnetic heading info is displayed.
> 
> one thing to note: in this version and versions prior to the rework on the RF driver, the calibration of my CC1101 board was really skewed to the low end of the spectrum:
> 
> `.*******.............................................................................................. Frequency = 869701.94kHz Range = 17.94kHz Deviation to real frequency = -138.06kHz Do you want to save the new RF calibration values (y/n) ? Configuration saved`
> 
> RF calibration is more "centered" with the latest code:
> 
> `.......................************************....................................................... Frequency = 869793.19kHz Range = 68.79kHz Deviation to real frequency = -46.81kHz (-53 ppm) Do you want to save the new RF calibration values (y/n) ? Configuration saved `
> 
> so maybe, somehow, my CC1101 is causing this...

This is normal, this is one of the issues fixed in the various pull requests. MicronetToNMEA was not setting the frequency correctly. The error was compensated by the frequency calibration but the rror in ppm was wrong. Now, your value of 53ppm is correct and really shows the error of your XTAL.