Rodemfr/MicronetToNMEA #55: SPI signal anomaly

dwarning, 2022-11-17:
I made few pictures with the actual master branch. I see a slight different behaviour of the MISO signal before the first traffic to CC1101.

This is an overview of the first transmission phase:

[image]

Next 3 pictures show a CS dependent High level of the MISO signal (don't know the meaning):

[image]
[image]
[image]

Now after 4.5ms follows the the correct MISO behaviour with my network ID:

[image]

Rodemfr, 2022-11-17:
The first SPI glitch you detected at startup is probably due to a bad initialization of the SPI pins in CC1101 driver. I will have a look to that tonight.
The second point is normal I think. Releasing CC1101's CS, means that CC1101 must stop driving all MOSI line to potentially let another IC talk on the bus. So CC1101 is switching to High-Z while Teensy is configured as input (High-Z also). So, the MISO line not beeing driven anymore, it takes an undefined value which depends on the High-Z circuitry of both Teensy & CC1101.

Rodemfr, 2022-11-17:
I pushed a possible fix in branch `fix_issue-55`. I'm not equipped to observe SPI signals right now. Could you check ?
I'm not sure it will really solve the issue, I have seen that SPI drivers is writing a number of HW registers related to SPI pins which I cannot remove or directly control.

dwarning, 2022-11-18:
Sorry, I am a bit lost. 

Branch fix_issue-55 works well.
The described behaviour above has not changed.
But if I go to CPU clock 24MHz (SPI clock stay same 4MHz)  your explained plausible High-Z behaviour disappears.
You have a specific interest to scope? 

BTW: Why you exchanged digitalWrite (startup phase) before pinMode? Looks unlogical.
Have exchanged it back for trial - it works OK.

Rodemfr, 2022-11-27:
I exchanged digitalWrite with pinMode to ensure that the value to output will be set before enabling output to avoid any transitory state. However, reading your message, I think I misunderstood the problem. Forget about this fix, it's out of purpose.
However, for MISO strange level when CS is high, I maintain that this is normal : when CS is high, the slave must release the line and go to high-Z, leading to unexpected values since there are not other slave to talk on the line. The effect of CPU speed is however strange and I can't explain it.

Rodemfr, 2022-12-02:
Looking again to your records, I think more and more that the ramping down MISO value is due to a fast transitory state of CS. The very first assertion of CS we see on your records is probably making CC1101 pull MISO high, but as CS immediately goes low, CC1101 releases control of MISO and switch it to high-Z. MISO then ramps slowly down until next assertion of CS which is for a more normal transfer at the end of which CC1101 has time to pull is down before releasing control.
Why do we have this spike ? I don't know. Since my previous fix did not change anything, maybe we have a CS change somewhere inside SPI driver initialization, or maybe is it a side effect of the reset sequence...
Anyway, I propose not to solve this one. I fear to pass a lot of time on it while it is not functionally problematic.

closed 2022-12-04 by Rodemfr