Rodemfr/MicronetToNMEA #59: Teensy 4.0 stops receiving packets

j-lang, 2022-11-28:
My MicronetToNMEA module was running during summer without any problems. 
I dismounted the PCB from the boat and tried to update to the current version, because there are some improvements. ([8c0c5be](https://github.com/Rodemfr/MicronetToNMEA/commit/8c0c5be780d65c9e8aef070455d8a7f419ac364c))
I build the code for a Teensy 4.1 and Teensy 4.0 board but could not receive Micronet data:
/!\ No Micronet network found /!\
Check that your Micronet network is powered on.
Calibrate RF frequency: is showing dots only.

After uploading the old code from beginning of july, (https://github.com/Rodemfr/MicronetToNMEA/issues/11) both boards where working well. 

Rechecked my settings in BoardConfig.h (GDO0: Teensy 4.0: pin 14; Teensy 4.1 pin 24) GNSS: Serial1 und WIRED NMEA: Serial 2:

```
#else // Teensy 4.0 Configuration
#define CS0_PIN  10
#define MOSI_PIN 11
#define MISO_PIN 12
#define SCK_PIN  13
#endif
#if defined(ARDUINO_TEENSY40)
    #define GDO0_PIN 14
#endif
#if defined(ARDUINO_TEENSY41)
    #define GDO0_PIN 24
#endif
```
# Problem
On Teensy 4.0 the LED is blinking 2 to 3 times and stops blinking, sometimes ist works for some minutes -> this allows to run "Calibrate RF frequency". But most of the starts are ending after a few seconds. - as soon as i install the july version, everything works well.
On Teensy 4.1 the LED is blinking and is connected to the Micronet network. 

No difference between 24MHz and 600MHz.
I am using Visual Studio Code with Platform IO. 

#Test
After installation of firmware downloaded at 10. of July: (0.11)
Starting Frequency tuning
Press ESC key at any time to stop tuning and come back to menu.

......................**************************......................................................
Frequency = 869793.19kHz
Range = 74.77kHz
Deviation to real frequency = -46.81kHz
Do you want to save the new RF calibration values (y/n) ?
Configuration saved

Choice : 2

Scanning Micronet networks for 5 seconds ... done

List of scanned networks :

Network 0 - 8xx (very strong)

MicronetToNMEA, Version
0.11
Device ID : 3xx
Attached to Micronet Network 8xx
RF Frequency offset = -46.81 kHz (-53 ppm)
**Working from now on** 

Update to firmware [8c0c5be](https://github.com/Rodemfr/MicronetToNMEA/commit/8c0c5be780d65c9e8aef070455d8a7f419ac364c)

Starting Frequency tuning
Press ESC key at any time to stop tuning and come back to menu.

................................................................***********************...............
Frequency = 869914.31kHz
Range = 65.80kHz
Deviation to real frequency = 74.28kHz (85 ppm)
Do you want to save the new RF calibration values (y/n) ?
Configuration saved

Choice : 2

Scanning Micronet networks for 5 seconds ... done

List of scanned networks :

Network 0 - 8xx (very strong)

MicronetToNMEA, Version 1.3
Device ID : 3xx
Attached to Micronet Network 8xx
RF Frequency offset = 74.28 kHz (85 ppm)
**I can start only one times " Start NMEA conversion" **
After pressing ESC key:
Starting Frequency tuning
Press ESC key at any time to stop tuning and come back to menu.

......................................................................................................
Choice :

No Idea what is wrong. 
On Teensy 4.1 with GDO0 on Pin 24 it works.

Rodemfr, 2022-11-28:
Changes have been made around frequency handling since july. Reading your logs I suspect a problem in the frequency tracking algorithm. Can you try to comment this line of Main.cpp (line 602) : 
`gRfReceiver.EnableFrequencyTracking(gConfiguration.networkId);`

Rodemfr, 2022-11-28:
Another point : I discovered that my previous calibration code was flawed and was indicating a wrong frequency offset. It explains why july's version is indicating -46.81kHz while the new one shows 74.28kHz

j-lang, 2022-11-29:
I commented out line 602, but no change. 
What could be the difference between Teensy 4.0 and Teensy 4.1 isr? Teensy 4.1 with GDO0 on Pin 24 ist working and Teensy 4.0 with GDO0 on Pin 14 does not work with the newer code versions. 
Another observation on Teensy 4.0: when I upload the "old" firmware the receiver works. After uploading to "new" firmware, the first run frequency tuning works, a second start has only dots. If uploading "new" firmware again, has no change -> only dot. But if I repeat the procedure with upload "old" firmware, start frequency tuning and then upload "new" firmware the frequency tuning runs only one times successfully.
I am  out for business for two day and will check things again afterwards.

Rodemfr, 2022-12-01:
There is something you can test : at line 170 change line
`attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, HIGH);`
by
`attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, RISING);`

I think there is still something wrong in the way IRQ are handled by MTN, but I don't know what.

Rodemfr, 2022-12-02:
I this last proposal does not change your issue, I have a last track to follow : I recently reduced SPI timings to speed up ISR processing. It might cause troubles of some devices. Let me know the result of the previous change.

j-lang, 2022-12-03:
I suppose you are right with "some devices":
1.: Change of line 170 had no impact on both boards:
`attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, HIGH);` by `attachInterrupt(digitalPinToInterrupt(GDO0_PIN), RfIsr, RISING);`
2.: The Teensy 4.0 is soldered into the PCB with the other components (schematic in [hardware/v1](https://github.com/Rodemfr/MicronetToNMEA/tree/master/hardware/v1)) that was mounted in my boat. Have a Teensy 4.1 for lab where I made my test with another CC1101 board. 
To crosscheck the Teensy 4.0 issue I got another Teensy 4.0 for lab and wired the CC1101 from Teensy 4.1 to the new Teensy 4.0 and it works with the "new" firmware. The CC1101 soldered the the PCB only works with the old version prior to July.
Additional info: the PCB version has power supply over USB port of the Teensy 4.0. For crosschecking I removed Recom power regulator and disconnected the GPS module. -> Only the Teensy 4.0 and the CC1101 are connected on the PCB. Unfortunately I have no spare CC1101 and have to wait till another CC1101 will arrive.

"good" CC101:
[image]

"bad" CC1101:
[image]

Rodemfr, 2022-12-03:
This CC1101 is obviously more sensitive to something in MTN code. You can try to checkout commit `c4c9692` (11/11/2022). This is the commit before I reduced some timings in SPI to go faster. If this commit works well, it means that this CC1101 is I some way slower than mine and that I have to increase MTN's timings.

j-lang, 2022-12-03:
With commit c4c9692 the same one works, the other not, and made a check with version from 10th of July, this version works still with both CC1101.

Rodemfr, 2022-12-03:
Damn ! I will check other modifications...

Rodemfr, 2022-12-03:
I don't find obvious modifications which could explain your issue. Would it be possible for you to make some sort of dichotomy with the commits between 10/07/2022 and today on master branch to know where the problem appears ? It is difficult for me to find this by myself since my CC1101 is OK.

j-lang, 2022-12-04:
Added output for CC1101 chip version (https://jgromes.github.io/RadioLib/class_c_c1101.html#a2a4c6e622dffd2788d8ac52d708b0705) 
"bad" chip has version: 4, 
"good" chip has version: 0x14.

dwarning, 2022-12-04:
Interesting: I can confirm Version 0x04 didn't working with latest  code. My other boards Version 0x14 are OK.

Rodemfr, 2022-12-04:
Interesting ! The fact that version of 10/07/22 is working and not the latest shows that there is something which disturbs CC1101 v04. I only have version 14. Someone to go through a dichotomy of commits between 10/07/22 and latest ?

j-lang, 2022-12-05:
I already checked some updates between 10/07/22 and 20/01/22 (my downloads) code parts of src\CC1101Driver.cpp and src\RfDriver.cpp without results. Are there other parts they setup the communication to CC1101?

Rodemfr, 2022-12-05:
I don't see how other files could play a role wrt CC1101, but I'm not sure. There was a lot of changes since july...

dwarning, 2022-12-05:
@j-lang : which version is the latest working with version 0x04? I see five commits on 10th July.
Will try to train my git skills for for- and backward winding.

dwarning, 2022-12-05:
Made few trials but I think it is not easy to playback because beside driver code also configuration and BoardConfig.h has changed.
It is easier to by an newer module and pray to get 0x14.

Rodemfr, 2022-12-12:
I would like to solve this issue before officially releasing MTN to avoid having people building a MTN setup which will not work.
Unfortunately, I have no 0x04 CC1101 to make the test myself. I really need one of you to track the commit which caused the compatibility problem. I know it is not a fun job, but this would really help.

j-lang, 2022-12-13:
Currently I have no idea which code parts are additionally to be checked. 
The CC1101 errata shows no difference between chip types: https://www.ti.com/lit/pdf/swrz020

j-lang, 2022-12-13:
I have no idea where to start.
TI errata shows no difference: https://www.ti.com/lit/pdf/swrz020
But there are some posts, they assume the maybe issue:
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/428028/cc1101-version-0x14-chip-difference
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/389977/wor-broken-on-version-0x14-parts

Rodemfr, 2022-12-13:
I have also no idea of what could be the difference.
My idea is that since you found a working commit (10/07/2020) and since we know the last commit does not work, to try to advance in the commits from 10/07/2020 to now, until you find the failing commit. This is not a fun job but I see no other option. To go faster you can advance 10 commits by ten commits, or to use dichotomy. There will be some trap since BoardConfig.h sometimes changed...

j-lang, 2022-12-13:
Got today some ordered CC1101 868 MHz boards but they have 0x04 chips. 
If I can send you one, please send me your address via e-mail.

dwarning, 2022-12-13:
I believe I found it. My setup is T4.1 / 04 CC1101.
The point is between 2f1f837 (only one times transmit) and ff28352 (working continuous).
It is the introduction of the so named low power mode. But the failing commit is very complex. So I am not able to investigate further.

Rodemfr, 2022-12-13:
Excellent ! I will dive into these commits. I'm out tomorrow, so I won't look into it before Thursday

Rodemfr, 2022-12-13:
Are you sure that `2f1f837` is failing while `ff28352` is working ? Because `2f1f837` is chronologically before `2f1f837`

dwarning, 2022-12-13:
Using gitk I see 2f1f837 is from 9:17 and ff28352 is from 10:21.
Yes my description is wrong: The older one worked and the newer one shows the failure. Sorry.
But I am not sure. Maybe @j-lang can make the same move.

j-lang, 2022-12-13:
Checked also https://github.com/Rodemfr/MicronetToNMEA/commit/2f1f83700b0105b4833f99e2b17023877ec5d141 and https://github.com/Rodemfr/MicronetToNMEA/commit/ff28352cf9623603b3145c01cf9927507933e086 and confirm that 2f1f837 can receive the Micronet  Network. 
By the way: the TeensyTimerTool library has a new version 1.2.0 without warnings. 

ff28352: I noticed some different things:
1. after Calibrate RF frequency (Menu 6) and Start NMEA conversion (Menu 4) the board resets after around 10 seconds in a loop. After then it resets also after a power up, but not always after power up.  
2. Calibrate RF frequency works and Micronet Network is recognized.
3. But the display show not updated time when the board does not reset in a loop.

I will make some more checks tomorrow for validation of this points.

Rodemfr, 2022-12-14:
I'm about to leave for business trip, so I have not much time to investigate but there is something you can test : on the latest commit of the master, empty the functions CC1101:LowPower() and CC1101:ActivePower(). If this is an issue regarding Low Power feature, it should work.

dwarning, 2022-12-14:
Good idea! I commented out the content of both functions, made a new calibration and started - now it works with latest git state.
Have a good business trip. Thanks too.

dwarning, 2022-12-14:
@j-lang : Can you please check only to rise the XTAL_RESTART_TIME_US (CC1101Driver.cpp) to 400. For me the 0x04 is working continuous with actual git master branch.

j-lang, 2022-12-14:
> @j-lang : Can you please check only to rise the XTAL_RESTART_TIME_US (CC1101Driver.cpp) to 400. For me the 0x04 is working continuous with actual git master branch.

Can confirm that it works by changing value from 150 to 400 with 600MHz and 24MHz with [847b544](https://github.com/Rodemfr/MicronetToNMEA/commit/847b54460f131e544d546f96ef7926424ab168c3) 

Great!

dwarning, 2022-12-14:
So this effect is more caused by crystal performance including layout and parasitic effects. Perhaps the package changes of version 04 vs 14 come in play. Rising the waiting time for start-up is also described in the cc1101 errata.
@Rodemfr : Can we live with this more as doubled duration? Or should we try to squeeze the value down. Alternative to look for XO ready would be monitoring for MISO rise. If I understood the spec. correct.

Rodemfr, 2022-12-14:
I did not read the errata, this explains the error.We can live with this value of 400us, its probably safer to let a big value here. I just need to adapt the timing for low power/active power accordingly. I will prepare a branch to test it.

Rodemfr, 2022-12-14:
Finally, there is no need for a branch : the timing was already OK for a 400us startup, I changed the value directly in master branch. Just for explanation : MTN set a timer at the end of each Micronet timer cycle (one every second) to put CC1101 in low power. Another timer is set 1ms before the start of the next network cycle to restart CC1101. 1ms is enough for a 400us XTAL startup and the various following configuration (a few hundred microseconds).

closed 2022-12-14 by Rodemfr