Rodemfr/MicronetToNMEA #60: Add support for automatic magnetic variation

Rodemfr, 2022-12-04:
Following discussion with @tvr256 : 
> Thinking about it some more, an even better solution would be for MTN to read both variation and deviation from Micronet and include them in the NMEA HDG sentence.
> 
> However, the TackTick documentation states that the variation setting "is ignored if variation information is available from a GPS source". I don't think the M8N sends variation in its RMC sentence (mine's not wired up so I can't check), but the ultimate solution would be for MTN to calculate variation based on M8N lat/long if it's available, using [a similar algorithm to SignalK](https://github.com/SignalK/signalk-derived-data/blob/master/calcs/magneticVariation.js), and then add it to the RMC and HDG sentences. Or would that be complete overkill?

I confirm M8N does not provide magVar. Calculating magnetic variation with [WMM from NOAA](https://www.ngdc.noaa.gov/geomag/WMM/) seems easy and this looks really appealing. However, I'm a bit worried about EEPROM handling. Each time a new variation would be calculated, different from the previous one, MTN would send an update to all Micronet devices which would burn it into their EEPROM. This could dramatically reduce EEPROM life span, especially if you sail near a magVar transition. Also, MTN will have to behave differently whether a GPS is connected or not, which will make MTN configuration even more difficult than today. We still have the possibility to add WMM calculation only for HDG sentence and to keep the possibility of some inconsistency between NMEA's HDG Micronet's HDG display. That's a possibility.

Concerning magnetic deviation, there is a menu in my T111 to handle deviation ("dev" in Option->Compass). When I select it, it displays "wait", then cancels. There is probably a procedure between display and compass to calibrate the system by making a slow 360° with the boat. I will spy the network during this procedure to see if I can find clues on it.

tvr256, 2022-12-04:
You make a good point about EEPROM handling. My interpretation of the TackTick documentation is that it only stores the user-configured Variation setting in EEPROM, and any variation received in an HDG or RMC sentence would be stored non-persistently in RAM. But that's just a guess, I may be wrong.

I see exactly the same as you when I access the Deviation menu, so I'm afraid I can't help there. The TackTick documentation states :

```
Press the > button to advance to the Deviation Page.
Press the O button to enter Compass Calibration Mode. Turn the vessel slowly keeping the speed below 4 knots and taking approximately 2 minutes to complete 360 degrees. Keep turning the vessel until the display changes to show a value (usually about 1.25 turns).
If the rate of turn of the vessel is too fast the display will show “TURN HIGH”. It is not necessary to abandon the turn at this stage but do lessen the rate of turn by reducing speed or widening the turning circle.
```

Rodemfr, 2022-12-04:
For deviation, there is probably a protocol between the display and the compass to handle the sequence. I would need a microcompass or a race master to spy it and reverse engineer it.
What you wrote for magnetic variation is very interesting. There might be a Micronet message dedicated to provide magVar from an external RMC sentence from a GPS in which case we can imagine that Micronet devices do not burn this one in EEPROM and just keep it in RAM for some time. If we could get the definition of this message we could integrate WMM model in MTN and use it for both RMC and Micronet without risking killing the EEPROM. We need a T122 NMEA interface and something to generate RMC with magVar to spy the traffic.

tvr256, 2023-03-06:
I'd be keen to test this, and I have a T122 NMEA interface on the boat. What's the best way to spy the traffic?

Rodemfr, 2023-03-07:
You have a T122 : excellent !

Spying is easy : just bring your laptop and MTN with you a go to menu `5 - Scan surrounding Micronet traffic`
All the traffic will be printed to the console.
There are several things which are interesting to look at :
- Magnetic Variation : It might be that T122 sends a unknown command when HDG sentence is sent to it with variation field filled. A command different of the *SetParameter(0xff 0x0d)* which triggers an EEPROM write.
- What is the MSB of the Device ID of a T122 ? This MSB is the device type and since I never crossed the path of a T122, I don't know it.
- A more tricky one : According to T110 (single display), it is capable to display performance data (polar speed?). The only way I see for Micronet to receive this data is via T122 or via Microtalk (Seatalk NG/Micronet bridge). Raymarine has a proprietary NMEA sentence to give this type of data. Could it be that a recent T122 could decode this sentence ? I will go and search information on this Raymarine proprietary NMEA sentence, I know qtVlm can send it...

Rodemfr, 2023-03-07:
If you want to go through the third item of the list, here is some information. I played a bit with qtVlm to see which NMEA sentences are output when enabling Raymarine Performance Data :

First I see a initial burst of `$PTAK` sentences which are sent only once and which seem to define what are the 6 parameters which will be sent periodically :

```
$PTAK,FFP1,%POLAR,   %*23
$PTAK,FFP2,TSPEED,   %*56
$PTAK,FFP3,=ANGLE,  @*7D
$PTAK,FFP4,=TACK,   @*06
$PTAK,FFP5,STREAM,  KTS*17
$PTAK,FFP6,STREAM,   @*38
```

Then periodically, `$PTAK` is used again to send the data :

```
$PTAK,FFD1,123*4B
$PTAK,FFD4,320*4F
$PTAK,FFD3,175*4A
$PTAK,FFD2,172*4C
$PTAK,FFD5,0*4F
$PTAK,FFD6,130*4E
```

It seems pretty simple to understand how it works.

tvr256, 2023-03-07:
My NMEA sender is wired into the boat, I'll be on board after Easter so I'll see what I can do.

I looked into the PTAK sentences some time ago, TackTick supposedly published a reference manual but I've never been able to find it. But from what I can piece together, the following displays can display a custom "Performance" chapter:

T110 single display supports 4 x custom pages
T113 remote display supports 6 x custom pages
T210 Maxi single display supports 6 x custom pages

Unfortunately the T215 Maxi dual display (and presumably the T111 dual display) don't support custom pages:
http://www.teamkashmir.com/manuals/uu074_v04_gb_Maxi%20Display_qs.pdf

The FFP (page) sentence is transmitted once every 25 seconds, and defines the 6 character header and 6 character footer displayed on each page. The FFD (data) sentence is transmitted once a second, and defines the data to be displayed on the page. So decoding your sentences:

```
$PTAK,FFP1,%POLAR,   %*23        Page 1 header "%POLAR", footer "   %  " 
$PTAK,FFP2,TSPEED,   %*56          Page 2 header "TSPEED", footer "   %  "
$PTAK,FFP3,=ANGLE,  @*7D        Page 3 header "ANGLE ", footer "  °   "
$PTAK,FFP4,=TACK,   @*06         Page 4 header "TACK  ", footer "   °  "
$PTAK,FFP5,STREAM,  KTS*17        Page 5 header "STREAM", footer "  KTS "
$PTAK,FFP6,STREAM,   @*38        Page 6 header "STREAM", footer "   °  "
```
As well as sending data to be displayed, there's also a `$PTAK,TWC,nn.n,nnn*hh` sentence which sends a True Wind Correction value, to correct for upwash.

Rodemfr, 2023-03-08:
No problem for the time frame. I will also be onboard in the same period to make real life test of MTN on my boat.
I will create a new issue to split magnetic variation and performances investigations.

closed 2025-11-15 by Rodemfr