Rodemfr/MicronetToNMEA #62: Undocumented Micronet sentence from wind transducer

tvr256, 2022-12-04:
When my Micronet system starts, I receive multiple `E2 09` sentences from my wind transducers (yes, there are 2 wind transducers in range) and an `F9 09` from the network. I can't find either in the message spec, are they new/useful?

```
810801CB 02036E47 06 01 09 1F 10 10  -- E2 09 01 EC  (-44, 36871)
810801CB 02036E47 06 01 09 1F 10 10  -- E2 09 01 EC  (-44, 36878)
810801CB 02036E47 06 01 09 1F 10 10  -- E2 09 01 EC  (-44, 36885)
810801CB 02036E47 06 01 09 1F 10 10  -- E2 09 01 EC  (-44, 36650)
810801CB 02036E47 06 01 09 1F 10 10  -- E2 09 01 EC  (-44, 36659)
810801CB 02076D31 06 01 09 0C 10 10  -- E2 09 02 ED  (-37, 28663)
810801CB 02076D31 06 01 08 0B 10 10  -- E2 09 02 ED  (-37, 28680)
810801CB 02076D31 06 01 08 0B 10 10  -- E2 09 02 ED  (-37, 28695)
810801CB 02076D31 06 01 08 0B 10 10  -- E2 09 02 ED  (-37, 28710)
810801CB 02076D31 06 01 08 0B 10 10  -- E2 09 02 ED  (-37, 28727)
810801CB 810801CB 06 01 00 B1 10 10  -- F9 09 03 05  (-50, 20311)
```
I also get a "WIND BATT" warning message on my master display at startup, could this be related?

Full log is attached, I haven't decoded every sentence so there may be other useful messages too.

[full log at startup.txt](https://github.com/Rodemfr/MicronetToNMEA/files/10149747/full.log.at.startup.txt)

Rodemfr, 2022-12-05:
I think you are right on the nature of the message.
I can provoke a low battery warning on my hull transmitter. I will do this and compare the message to yours. I will first finish fixing the other issues before working on this one.

Rodemfr, 2023-03-20:
I advanced on this topic. I let my hull transmitter go to low battery and logged the related messages.
The message ID used to send the alarm is 0x06 and the payload (without the last CRC byte) is :

`E2 08 04` -> Low battery alarm from hull transmitter
`F9 08 05` -> Stop/acknowledge alarm from display

doing the test again I get :

`E2 08 01` -> Low battery alarm from hull transmitter
`F9 08 02` -> Stop/acknowledge alarm from display

I also observed the following payload when the hull transmitter is not powered (no volot alarm):

`F9 0C 01` -> Clear no volt (ACK expected)

So, adding this to your own observation, we can conclude the following : 
- The first byte of the payload is the command
  - 0xe2 to trigger an alarm
  - 0xf9 to stop/ack it
- The second byte is the alarm ID : 
  - 0x09 for low batt from wind tranducer
  - 0x08 for low batt from hull transmitter
  - 0x0c for no volt from hull transmitter
- The last byte seems to be a sequence number. A number incremented by one at each new alarm related command (0xe2 or 0xf9). I'm unsure if this number is proper to each type of alarm or if it is global to all types of alarm.

I will do some test to try to discover all types of alarms by sending fake 0xe2 commands. Then I will update the spec.

Rodemfr, 2023-04-10:
Documentation updated.