Rodemfr/MicronetToNMEA #64: Data precision in HDG and VHW sentences

tvr256, 2022-12-11:
According to the NMEA spec, heading, variation and water speed should all be reported to 1 decimal place in the NMEA HDG and VHW sentences. Currently MicronetToNMEA uses 0 decimal places for heading and variation, and 2 decimal places for water speed.

My Git skills are non-existent so I won't try to provide a patch. But the following fields should be changed in DataBridge.cpp:

Line 813: `"$INVHW,,T,%.0f,M,%.2f,N,,K"` should read `"$INVHW,,T,%.1f,M,%.1f,N,,K"`
Line 817: `"$INVHW,,T,%.0f,M,,N,,K"` should read `"$INVHW,,T,%.1f,M,,N,,K"`
Line 821: `"$INVHW,,T,%.0f,M,,N,,K"` should read `"$INVHW,,T,%.1f,M,,N,,K"`
Line 842: `"$INVHW,,T,,M,%.2f,N,,K"` should read `"$INVHW,,T,,M,%.1f,N,,K"`

Also a very minor point - the comments in BoardConfig.h refer to the wrong NMEA sentence names:

`(SPD, LOG)` should read `(VHW, VLW)`
`(XDG)` should read `(XDR)`
`(STP)` should read `(MTW)`

Rodemfr, 2022-12-12:
Easy to fix, will be done tonight.

Rodemfr, 2022-12-12:
Done & tested with OpenCPN

closed 2022-12-12 by Rodemfr