Rodemfr/MicronetToNMEA #65: Add depth offset to NMEA DPT sentence

tvr256, 2022-12-11:
Currently MTN always reports a depth offset of 0.0 in the NMEA DPT sentence (ie it always reports transducer depth)

There's a setting in Micronet to input the depth offset, which allows you to set depth below surface or depth below keel. Would it be possible to capture this (in the same way as we capture compass offset) and add it to the DPT sentence?

Rodemfr, 2022-12-12:
It is possible. It will be done tonight.

Rodemfr, 2022-12-12:
Done. Tested with OpenCPN : transducer offset is correctly interpreted by OCPN.

tvr256, 2022-12-13:
I don't have my depth sensor at home so I can't test, but reading the code it looks like an extra comma might have been added to the end of the sentence by mistake?

Rodemfr, 2022-12-14:
`"$INDPT,%.1f,%.1f,"`

tvr256, 2022-12-14:
You're absolutely correct. Apologies - I was only comparing against the previous DPT readout, I didn't check the spec.

There are a few unofficial versions of the NMEA spec floating around on the internet, some of which contradict each other. The only official one I've found is v3.0.1 from 2002 : https://www.plaisance-pratique.com/IMG/pdf/NMEA0183-2.pdf And this definitely includes the extra range scale parameter.

Rodemfr, 2022-12-15:
An official spec of NMEA ! I downloaded it. I will use it as reference now and review all the sentences encoded/decoded by MTN.

closed 2022-12-15 by Rodemfr