Rodemfr/MicronetToNMEA #72: Water speed displayed when no speed sensor connected

tvr256, 2022-12-16:
My hull sender and water speed transducer aren't connected, but for some reason my TackTick display shows water speed updating with the same value as SOG.

Is MTN accidentally updating the water speed as well as SOG? Or is the TackTick display using SOG for water speed if the speed sensor isn't connected?

[image]

dwarning, 2022-12-16:
Which git status you are using?
With latest commit from today.
I can't confirm your report. My display showing "----" for both.
But something seems changed. In the past I got low SOG values (e.g. 0.2) which I assume comes from noisy data from GNSS. Now SOG shows nothing, what is not wrong but new. Also make some move with GNSS did not show SOG.

tvr256, 2022-12-16:
I'm using the latest commit from today. If you're getting "---" for SOG, I think that just means you don't have a GPS fix.

dwarning, 2022-12-16:
No, I have a fix - lat and long is there. But if I am not moving gps has no value for sog/cog. In longer time gps catch some noise and show a low and wrong value. This is now happen after some minutes. Speed is still "----".
So, I think on my side everything is OK.

Rodemfr, 2022-12-16:
I don't observe the same : my two tacktick T111 are displaying SOG from M8N but nothing for water speed. My first T111 is a white "tacktick" like you. In the health setup page it shows FW version 29. My second one is a newer gray "Raymarine" showing version 37.2.
My BoardConfig/NMEA config is :
```
#define NAV_SOURCE_LINK     LINK_NMEA_EXT  // Navigation data (RMB)
#define GNSS_SOURCE_LINK    LINK_NMEA_GNSS // Positionning data (RMC, GGA, VTG)
#define WIND_SOURCE_LINK    LINK_MICRONET  // Wind data (MWV)
#define DEPTH_SOURCE_LINK   LINK_MICRONET  // Depth data (DPT)
#define SPEED_SOURCE_LINK   LINK_MICRONET  // Speed data (VHW, VLW)
#define VOLTAGE_SOURCE_LINK LINK_MICRONET  // Battery voltage data (XDR)
#define SEATEMP_SOURCE_LINK LINK_MICRONET  // Temperature data (MTW)
#define COMPASS_SOURCE_LINK LINK_COMPASS   // Heading data (HDG)
```
Could it be that your config receives speed from NMEA ?

Rodemfr, 2022-12-16:
An idea for investigation : you can look what MTN outputs as water speed on NMEA. If not bugged, it should not output any valid water speed.

tvr256, 2022-12-16:
My config is the same as Ronan's so I don't think I'm getting speed from NMEA. And the VHW sentence from MTN doesn't contain any water speed.

My T111 has firmware version 24, it has a 2005 date on the back so I think it's one of the earliest ones. So it could well be a firmware bug. If no-one else is seeing the same thing, then it's probably not an MTN issue.

Rodemfr, 2022-12-16:
That's also what I think : a bug or maybe just a bad initial idea to duplicate SOG on Water speed when it is not available.

closed 2022-12-17 by Rodemfr

tvr256, 2023-02-22:
Confirmed - it's a bug in T111 firmware 24.

I've tried a different T111 with firmware 35.4, and SOG isn't duplicated to the Speed display:

https://photos.google.com/photo/AF1QipN0dXMBpOm_8Hbe0-SnRVrxt3cpzTuRcT_ySuFk