Rodemfr/MicronetToNMEA #84: Bootloop with Teensy 4.18.0 / 1.58

j-lang, 2023-05-21:
There is an issue with the current Teensy package ver. 4.18.0 for PlatformIO.
I updated VSCode and platforms. Afterwards made a configuration change in BoardConfig.h and uploaded new code.
Result: orange LED is on, and every 8 second the board starts, windows shows and unknown usb serial device. 
Undo of last change -> LED is on every 8 seconds reboot of teensy.
Tested the same on a second Teensy 4.0 -> the same problem. 
Tried to build with Arduino IDE (everything up to date) -> the same problem
Tried everything on another PC -> the same problem (fresh installation of PlatformIO and Arduino IDE)
"Blink" example code works on every Platform.
After hours I tried installing Teensy Package 4.17.0 that worked in march -> the code is running again!

Currently not working package: Arduino Teensy 1.58.0, 1.58.1, PlatformIO 4.18.0
Currently latest working package: Arduino Teensy 1.57.2, PlatformIO 4.17.0

Rodemfr, 2023-05-21:
I have the same problem. No idea where it comes from so far. I will investigate...

Pacer27, 2023-05-22:
Hello, this Problen is known.

https://github.com/platformio/platformio-core/issues/3742

Rodemfr, 2023-05-23:
Thanks for the link, I will check the workaround.
Until then I confirm that switching back to PIO 4.17 solves the issue. I will study the PIO workaround in Pacer27' link and try to understand why Blink works while MTN is not even able to write a simple text a loop.

Rodemfr, 2023-06-11:
The problem really seems to come from Teensy's SW stack. And there is no easy way for MTN to avoid it. So, until the problem is fixed, I added a warning in the homepage readme to use Teensyduino 1.57.2

nmeyne, 2023-12-11:
Hi!   I'm not sure if this is related to the same, or a different stack issue, but I got compile errors with the Arduino IDE v2.2.1 when using Teensyduino 1.57.2 and TeensyTimer Tool version 1.4.1.  I regressed to TTT v 1.3.1 and it all compiled and ran fine.

Rodemfr, 2025-01-11:
The problem has been solved with latest versions of PlatformIO. THe initial bug was likely in Teensyduino package.

closed 2025-01-11 by Rodemfr