Rodemfr/MicronetToNMEA #95: Add Support for HMC5883L Compass

rschoolm, 2025-02-24:
Hello and thank you for all your efforts.

I don't know if I am asking for a lot, but I was wondering if support can be added for the HMC588L Compass.  I bought this GPS and found that it has a compass built into it.

See here: M8N GPS Module Built-in Compass ( https://www.amazon.com/dp/B06XR6JHDW?ref=ppx_yo2ov_dt_b_fed_asin_title )

Rodemfr, 2025-03-03:
Hello rschoolm,

This is unlikely that a driver for HMC588L will come in MTN. This would require me to acquire the hardware, write the code and test it. I lack time to do this. technically, if someone comes with a pull request for it, I could integrate it but would be unable to test it and maintain it.

Best regards,

Ronan.

rschoolm, 2025-03-08:
Understood.  Thank you for all your efforts on the current project put forth.